smartemission/smartemission #6: eenvoudige download

geurp0, 2016-03-08:
Vraag om een eenvoudige download per sensor. enkele knop, met download laatste 24 uur of download laatste week.

pietermarsman, 2016-06-02:
Wordt ook naar gevraagd op het forum: http://smartemission.ruhosting.nl/forums/topic/deelnemer-wil-graag-de-historie-kunnen-opvragen/

justb4, 2016-07-18:
In testfase is download via WFS: http://test.smartemission.nl/heron. Klik op verrekijker icoon en kies/station en sensor. In principe is het kwestie WFS url, bijv
http://test.smartemission.nl/geoserver/wfs?request=GetFeature&typename=timeseries_no2&outputformat=csv&CQL_FILTER=device_id=20 (alle NO2 voor station 20). Iemand zou daar op website eenvoudiger versie van kunnen maken. Data platform biedt in ieder geval de mogelijkheid.

Tevens ook via SOS of SOS REST (ook in testfase), zie bijv de 52North viewer:
http://test.smartemission.nl/sos52n/static/client/jsClient/index.html

Zie alle links ook op http://test.smartemission.nl. Binnenkort ook op productie.

justb4, 2016-07-18:
NB via zoeken "Door geometriën te tekenen" (zie Heron viewer verrekijker icoon, dropdown 2e zoekscherm) kunnen zelfs RIVM historische waarden, bijv Ruyterstraat sinds 2014 worden opgehaald....Is wel intensiever, langere wachttijden.

justb4, 2016-08-22:
En de handleiding: http://smartemission.ruhosting.nl/forums/topic/handleiding-om-sensor-data-te-downloaden/

closed 2016-08-22 by justb4