smartemission/smartemission #10: Data sensoren maakt datumsprongen in WFS

bdg-intemo, 2016-07-20:
Sensor 41 begint bij 25 juni, terwijl in de Whale viewer ook vroegere data te vinden is. Sensor 12 bijvoorbeeld maakt een sprong van april naar juli, terwijl deze data ook gewoon in de Whale viewer te vinden is.

justb4, 2016-07-21:
Ja zie het ook. Veel begint op 25 juni, terwijl de Harvester vaak vanaf april al heeft "geoogst", m.a.w. de ruwe data is wel via de Raw Whale API opgehaald, maar daarna is niet alles door Refiner/Publisher gegaan. Kan bijv zijn dat de validatie te streng was of iets anders.
Het is wel zo dat veel van de gecalibreerde gassen ontbreken over de gehele tijdspanne. Meestal is dat omdat s_temperatureambient niet beschikbaar is.

bdg-intemo, 2016-07-21:
In het geval van sensor 41, waar ik dit door zag, zijn er gewoon temperatuur en gassen aanwezig. Enig ander idee waardoor het kan komen? Ik zou de data graag downloaden!

This message has been scanned for viruses and
dangerous content by MailScanner<http://www.mailscanner.info/>, and is
believed to be clean.

Please consider the environment before printing this email

This e-mail, and any files transmitted with it, is confidential and is intended solely for the use of the person or entity to whom it is addressed. If you have received this e-mail in error, please notify the sender and delete it from your records.

On all our offers, sales and deliveries our General Terms of Delivery are applicable, excluding all other Terms and Conditions. Our terms have been deposited at the Chamber of commerce Eindhoven (The Netherlands) august 23rd, 2001, reference number 17065041, a copy of these terms is enclosed with all quotations, a hardcopy also can be send on your request. You can find our General Terms online at http://www.intemo.com/en/about-intemo/terms-and-conditions-delivery/
Op al onze aanbiedingen, verkopen en leveringen zijn, met uitsluiting van alle andere voorwaarden, onze algemene voorwaarden van toepassing. Deze zijn gedeponeerd bij de Kamer van Koophandel te Eindhoven onder nummer 17065041 per 23 augustus 2001. Bij iedere offerte hoe ook verzonden zijn de algemene voorwaarden bijgevoegd, een kopie kan op aanvraag ook per post worden toegezonden. U kunt onze algemene voorwaarden online inzien op http://www.intemo.com/nl/over-intemo/voorwaarden/

justb4, 2016-07-24:
Is m.i. met bovenstaande commit opgelost. Zowel de WFS als de SOS hebben geen grote (wel soms kleinere) datumsprongen meer. Zie de Heron (http://test.smartemission.nl/heron/ bijv History lagen en WFS) en SOS (http://test.smartemission.nl/sos52n/static/client/jsClient/index.html) viewers.  
De laatste kan nu ook waarden van de RIVM SOS http://inspire.rivm.nl/sos/eaq/api/v1/ laten zien (in Map View, klik onder-midden op folder/map blauwe icoon (meeste links van de drie).

closed 2016-07-24 by justb4

bdg-intemo, 2016-07-26:
Bedankt!