smartemission/smartemission #12: Juiste eenheden/min/max waarden tbv validatie: check nodig

justb4, 2016-07-21:
In de `Refiner ETL` worden ruwe sensor-waarden omgezet/gekalibreerd (Gassen) naar nieuwe waarden. Zowel voor als na de omzetting worden input en output waarden gevalideerd op minimale en maximale waarden. De omzetting wordt niet gedaan als een van de input waarden buiten min-max ligt. Dit geldt ook voor de omgezette waarden. Om dit te structureren is er een configuratie-bestand waarin alle metadata per component staat, zowel voor ruwe sensor-indicators (meestal prefix `s_` of `v_`) als omgezette waarden. Bijv `s_temperatureambient` (milliKelvin) is een ruwe waarde en `temperature` (Celsius) een omgezette waarde. Deze staan in bestand
https://github.com/Geonovum/smartemission/blob/master/etl/sensordefs.py
, m.n. vanaf regel 76. De vraag is: klopt alle de metadata in dit bestand met name voor de `s_` en `v_` indicatoren? Iemand van Intemo zou dit beste kunnen nagaan.

justb4, 2016-07-21:
Als voorbeeld: veel waarden worden niet omgezet/gepubliceerd omdat indicatoren buiten de min/max range vallen. Zie bijv log output hieronder voor de Last Values ETL. De validatie wordt gedreven door min/max waarden in https://github.com/Geonovum/smartemission/blob/master/etl/sensordefs.py :

```
2016-07-21 12:01:01,281 RawSensorAPI WARNING device_id=8 sensor=temperature: invalid input for s_temperatureambient: detail=s_temperatureambient not present
2016-07-21 12:01:01,281 RawSensorAPI WARNING device_id=8 sensor=humidity: invalid input for s_humidity: detail=s_humidity not present
2016-07-21 12:01:01,281 RawSensorAPI WARNING device_id=8 sensor=pressure: invalid input for s_barometer: detail=s_barometer not present
2016-07-21 12:01:01,281 RawSensorAPI WARNING device_id=8 sensor=noiseavg: invalid input for ['v_audio0', 'v_audioplus1', 'v_audioplus2', 'v_audioplus3', 'v_audioplus4', 'v_audioplus5', 'v_audioplus6', 'v_audioplus7', 'v_audioplus8', 'v_audioplus9']: detail=v_audio0 not present
2016-07-21 12:01:01,281 RawSensorAPI WARNING device_id=8 sensor=noiselevelavg: invalid input for noiseavg: detail=noiseavg not present
2016-07-21 12:01:01,282 RawSensorAPI WARNING device_id=8 sensor=co2: invalid input for s_co2: detail=s_co2 not present
2016-07-21 12:01:01,282 RawSensorAPI WARNING device_id=8 sensor=co: invalid input for ['s_o3resistance', 's_no2resistance', 's_coresistance', 's_temperatureambient', 's_temperatureunit', 's_humidity', 's_barometer', 's_lightsensorbottom']: detail=s_o3resistance not present
2016-07-21 12:01:01,282 RawSensorAPI WARNING device_id=8 sensor=no2: invalid input for ['s_o3resistance', 's_no2resistance', 's_coresistance', 's_temperatureambient', 's_temperatureunit', 's_humidity', 's_barometer', 's_lightsensorbottom']: detail=s_o3resistance not present
2016-07-21 12:01:01,282 RawSensorAPI WARNING device_id=8 sensor=o3: invalid input for ['s_o3resistance', 's_no2resistance', 's_coresistance', 's_temperatureambient', 's_temperatureunit', 's_humidity', 's_barometer', 's_lightsensorbottom']: detail=s_o3resistance not present
2016-07-21 12:01:01,282 RawSensorAPI WARNING device_id=8 sensor=coraw: invalid input for ['s_coresistance']: detail=s_coresistance not present
2016-07-21 12:01:01,282 RawSensorAPI WARNING device_id=8 sensor=no2raw: invalid input for ['s_no2resistance']: detail=s_no2resistance not present
2016-07-21 12:01:01,282 RawSensorAPI WARNING device_id=8 sensor=o3raw: invalid input for ['s_o3resistance']: detail=s_o3resistance not present
2016-07-21 12:01:01,305 RawSensorAPI WARNING device_id=9 sensor=co2: invalid input for s_co2: detail=s_co2 not present
2016-07-21 12:01:01,306 RawSensorAPI WARNING device_id=9 sensor=co: invalid input for ['s_o3resistance', 's_no2resistance', 's_coresistance', 's_temperatureambient', 's_temperatureunit', 's_humidity', 's_barometer', 's_lightsensorbottom']: detail=s_o3resistance not present
2016-07-21 12:01:01,306 RawSensorAPI WARNING device_id=9 sensor=no2: invalid input for ['s_o3resistance', 's_no2resistance', 's_coresistance', 's_temperatureambient', 's_temperatureunit', 's_humidity', 's_barometer', 's_lightsensorbottom']: detail=s_o3resistance not present
2016-07-21 12:01:01,306 RawSensorAPI WARNING device_id=9 sensor=o3: invalid input for ['s_o3resistance', 's_no2resistance', 's_coresistance', 's_temperatureambient', 's_temperatureunit', 's_humidity', 's_barometer', 's_lightsensorbottom']: detail=s_o3resistance not present
2016-07-21 12:01:01,306 RawSensorAPI WARNING device_id=9 sensor=coraw: invalid input for ['s_coresistance']: detail=s_coresistance not present
2016-07-21 12:01:01,306 RawSensorAPI WARNING device_id=9 sensor=no2raw: invalid input for ['s_no2resistance']: detail=s_no2resistance not present
2016-07-21 12:01:01,306 RawSensorAPI WARNING device_id=9 sensor=o3raw: invalid input for ['s_o3resistance']: detail=s_o3resistance not present
2016-07-21 12:01:01,306 dboutput INFO query is INSERT INTO last_device_output (name,point,value_raw,altitude,value,label,unit,value_stale,time,device_name,unique_id,device_id) VALUES (%s,%s,%s,%s,%s,%s,%s,%s,%s,%s,%s,%s)
2016-07-21 12:01:01,347 dboutput INFO committed 5 records
2016-07-21 12:01:01,369 RawSensorAPI WARNING device_id=10 sensor=temperature: invalid input for s_temperatureambient: detail=s_temperatureambient not present
2016-07-21 12:01:01,369 RawSensorAPI WARNING device_id=10 sensor=humidity: invalid input for s_humidity: detail=s_humidity not present
2016-07-21 12:01:01,369 RawSensorAPI WARNING device_id=10 sensor=pressure: invalid input for s_barometer: detail=s_barometer not present
2016-07-21 12:01:01,369 RawSensorAPI WARNING device_id=10 sensor=noiseavg: invalid input for ['v_audio0', 'v_audioplus1', 'v_audioplus2', 'v_audioplus3', 'v_audioplus4', 'v_audioplus5', 'v_audioplus6', 'v_audioplus7', 'v_audioplus8', 'v_audioplus9']: detail=v_audio0 not present
2016-07-21 12:01:01,369 RawSensorAPI WARNING device_id=10 sensor=noiselevelavg: invalid input for noiseavg: detail=noiseavg not present
2016-07-21 12:01:01,369 RawSensorAPI WARNING device_id=10 sensor=co2: invalid input for s_co2: detail=s_co2 not present
2016-07-21 12:01:01,370 RawSensorAPI WARNING device_id=10 sensor=co: invalid input for ['s_o3resistance', 's_no2resistance', 's_coresistance', 's_temperatureambient', 's_temperatureunit', 's_humidity', 's_barometer', 's_lightsensorbottom']: detail=s_o3resistance not present
[... cut ...] sensorbottom']: detail=s_o3resistance not present
2016-07-21 12:01:04,130 RawSensorAPI WARNING device_id=828231 sensor=coraw: invalid input for ['s_coresistance']: detail=s_coresistance not present
2016-07-21 12:01:04,131 RawSensorAPI WARNING device_id=828231 sensor=no2raw: invalid input for ['s_no2resistance']: detail=s_no2resistance not present
2016-07-21 12:01:04,131 RawSensorAPI WARNING device_id=828231 sensor=o3raw: invalid input for ['s_o3resistance']: detail=s_o3resistance not present
```

pietermarsman, 2016-07-21:
Gaat het hier om de eenheden/min/max waarden van de ruwe data?

justb4, 2016-07-21:
Beiden: er is validatie van inputs meestal s_ vars en van de outputs/resultaten bijv o3.

lindacarton, 2016-07-28:
Ik had begrepen  dat de sensorspecificatie alleen tussen een bepaalde range een zekere betrouwbaarheid meegeeft (voor testen in labs), maar dat onder een minimumwaarde nog steeds gas-weerstandswaarden geregistreerd worden.  Het is dan ook een vraag in hoeverre je de min-max bandbreedte wil aanhouden die op de Jose sensorspecificatie sheet staat. Hier kan Bas, Janus of Antoine het best input over geven.  Dit zou per sensing component kunnen verschillen. Misschien is meenemen van de waarde handiger dan bij voorbaat weglaten: je kunt bij historische data-analyses altijd zelf nog absurd lage of hoge waarden wegfilteren. Nu creëer je mogelijk meer 'gaten' in de data dan noodzakelijk. Ook voor burgers is het volgens mij vervelender als er geen data 'doorkomt' op de viewer,  dan wanneer die buiten een bandbreedte ligt; ik heb al eens vragen gekregen waarom de sensor soms geen data toont op  de SmartApp. In de documentatie zou kunnen komen te staan binnen welke bandbreedte volgens de fabrikant van de betreffende sensing-component een bepaalde nauwkeurigheid (precisie, betrouwbaarheid) verwacht mag worden.

justb4, 2016-07-28:
> Ik had begrepen dat de sensorspecificatie alleen tussen een bepaalde
> range een zekere betrouwbaarheid meegeeft (voor testen in labs), maar
> dat onder een minimumwaarde nog steeds [...]
Is iets complexer: voor calibratie is bijv temperatuur nodig. Die is op 
veel sensoren niet aanwezig (bijv station 25, Lent). Daardoor kunnen 
geen gecalibreerde waarden berekend worden. De vraag over min/max ruwe 
waarden ligt bij Intemo via dit issue:
https://github.com/Geonovum/smartemission/issues/12
De ruwe waarden hebben minder "gaten" en zijn beschikbaar middels de 
lagen eindigend op "*raw".

pietermarsman, 2017-01-25:
Is hier al meer over bekend? 

Gebruiken we nu de min/max uit de sensorspecificatie? 

Ik ben ervoor om dit iets op te rekken.

justb4, 2017-01-25:
> Is hier al meer over bekend?
>
> Gebruiken we nu de min/max uit de sensorspecificatie?
Ja, heb de mannen van Intemo gevraagd, zij verwezen naar hun spec.
>
> Ik ben ervoor om dit iets op te rekken.
Ja ga je gang, zie sensordefs.py

lindacarton, 2017-01-25:
Pieter, jij bent als de chief kalibratie  degene die het beste kan inschatten of je het meenemen van ruwe meetwaarden  laat begrenzen door de vooraf opgegeven min/max bandbreedte uit de fabrieksspecs,  of dat je die bandbreedte breder laat en d.m.v. de kalibratieprocedure later een uitspraak te doen over betrouwbaarheid van de metingen, ook de metingen in het lage of hoge spectrum. 
Dus jouw voorstel volgen: Alleen metingen die geen meting zijn (bijv. 2^16=65536 of error codes) eruit halen en alle metingen met een lage betrouwbaarheid erin laten.

Overigens is een GI-minor student nu bezig met een klein onderzoekje, analyse van de (ruwe waarden van) Geluid uit de sensordata. Om te kijken of hij een andere aggregatie kan maken voor de historische waarden dan uurgemiddelden voor geluid. Voor emissies werkt het terugkijken van uurgemiddelden goed, voor geluid is dat minder toepasselijk, omdat alle kortstondige geluiden worden uitgemiddeld.   
Hij gebruikt de geluidswaarden van de Whale server, dus per episode van 11 seconden het gemiddelde, de minimum en maximumwaarde per frequentieband.  En bijvoorbeeld kijkend naar de middelste frequentieband, naar het aantal (count)  episodes  dat er pieken (max waarden) zijn waargenomen  hoger dan 70 dB.  En dan kijken of aggregatie in de tijd een beeld oplevert  dat correspondeert met eigen waarneming van geluid. Met verschillende aggregaties en varianten van drempelwaarden (definitie van geluidspieken)  of waargenomen frequentiebanden (toonhoogten). 
Zijn doel is te komen met een advies aan het smart emission projectteam en de burgermeetwerkgroep.

pietermarsman, 2017-01-26:
Okido, dan ga ik kijken of en waar er te strakke restricties op de data zitten.

pietermarsman, 2017-04-11:
Volgens Henk kloppen  de restricties op de voorspelde gas waarden (ug/m3).

closed 2017-04-11 by pietermarsman