smartemission/smartemission #14: Historische Data download issue: Testbevinding voor 'noiselevelavg' url voor WFS download

lindacarton, 2016-07-28:
Testbevindingen Historische Data beschikbaarstelling via Smart Emission data platform en Heron viewer: 

De tijdserie begint op 20 juli, 10 uur en loopt door tot 28 juli =ik dacht even dat ik alleen de laatste dagen kon terugzien=, maar naar beneden scrollend in de lange lijst volgt daarna  
de data van 1 april  tot en met  20 juli (van day 20160401, hour 19 t/m 20160720, hour 9). 
Dit lijkt me verder geen probleem,  is handig als in de documentatie staat dat de "oude" data ouder dan 20 juli 2016 onder de andere dataregels verschijnt. 
 
De directe WFS download werkt, voor no2 en 03 en co2 geprobeerd. Alleen voor 'noiselevelavg' krijg ik een foutmelding:  Na url manipulatie met noiselevelavg voor bijvoorbeeld sensorstation 58, via deze url ....
http://data.smartemission.nl/geoserver/wfs?request=GetFeature&typename=timeseries_noiselevelavg&outputformat=csv&CQL_FILTER=device_id=58
... krijg ik deze foutmelding: 

<?xml version="1.0"?>

-<ows:ExceptionReport xsi:schemaLocation="http://www.opengis.net/ows/1.1 http://data.smartemission.nl:80/geoserver/schemas/ows/1.1.0/owsAll.xsd" version="2.0.0" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:ows="http://www.opengis.net/ows/1.1" xmlns:xs="http://www.w3.org/2001/XMLSchema">

-<ows:Exception locator="typeName" exceptionCode="InvalidParameterValue">

<ows:ExceptionText>Feature type :timeseries_noiselevelavg unknown</ows:ExceptionText>

</ows:Exception>

</ows:ExceptionReport>

justb4, 2016-07-28:
Ja de data is niet gesorteerd, is mogelijk WFS optie. Dus in Excel moet je op kolommen dag/uur sorteren.

Downloaden gemiddeld geluids nivo moet laagnaam: ``timeseries_noise_level_avg`` zijn (ipv  ``timeseries_noiselevelavg``). Stond niet goed in documentatie, nu wel. Dus de directe download URL is 
http://data.smartemission.nl/geoserver/wfs?request=GetFeature&typename=timeseries_noise_level_avg&outputformat=csv&CQL_FILTER=device_id=58

De "value" kolom is gemiddeld geluidsniveau (1-5) over uur gerekend. "raw_value" is ook uurgemiddelde maar in Decibel.

closed 2016-07-28 by justb4

lindacarton, 2016-07-28:
OK!

Heb je zelf al eens gekeken wat die historische data nu “laten zien” aan patronen?
Ik heb dinsdag aanstaande, 2 augustus, om 10:00 uur overleg met Freek, over use cases.

‘k ga nu naar huis,

Groet, Linda.

http://data.smartemission.nl/geoserver/wfs?request=GetFeature&typename=timeseries_noise_level_avg&outputformat=csv&CQL_FILTER=device_id=58

lindacarton, 2016-07-28:
PS. Is het er klaar voor om aan te kondigen aan de burgers?  En wil je dat zelf doen, op het Forum? Dat lijkt me wel de mooiste manier.
Dan bedoel ik het forum voor burgers, hier: http://smartemission.ruhosting.nl/?forum=forum

Is het handig als ik er ook een E-nieuwsbrief-mail voor burgers van maak/samenstel?
Met enkele zaken die op het Forum worden aangekondigd, met berichten van jou en Henk Nijhuis:  zoals de oproep van Henk Nijhuis en jouw  post over de beschikbaarheid van de historische data.  Het werken met de verrekijker lijkt me voor de gemiddelde burger het meest geschikt. (of de 52 North viewer)
Of is het nog niet zover en wil je eerst meer begeleidende informatie erbij leveren, zoals bijvoorbeeld handleiding…

Groet, Linda.

From: Carton, L.J.
Sent: donderdag 28 juli 2016 19:50
To: 'Geonovum/smartemission'; Geonovum/smartemission
Cc: Author
Subject: RE: [Geonovum/smartemission] Historische Data download issue: Testbevinding voor 'noiselevelavg' url voor WFS download (#14)

OK!

Heb je zelf al eens gekeken wat die historische data nu “laten zien” aan patronen?
Ik heb dinsdag aanstaande, 2 augustus, om 10:00 uur overleg met Freek, over use cases.

‘k ga nu naar huis,

Groet, Linda.

http://data.smartemission.nl/geoserver/wfs?request=GetFeature&typename=timeseries_noise_level_avg&outputformat=csv&CQL_FILTER=device_id=58

lindacarton, 2016-07-28:
Ik denk dat er idd nog documentatie nodig is. Volgens Michel zou 
Matthijs daar ook aan werken. Vraag is waar dat moet komen te staan: 
platform docu, website, etc. Ook is er in feite, komt denk ik ook door 
vakanties, nauwelijks door project-team getest (behalve door jou en 
Bas). Laten we eerst wat oneffenheden weghalen en de documentatie op 
orde maken. Anders is het gebodene nogal verwarrend kan ik mij voorstellen.

Er zijn bijv ook mogelijkheden nauwelijks bekend: bijv in de SOS Viewer 
kunnen ook RIVM uurwaarden worden getoond tezamen in grafiek en download 
met SE-waarden (via keuze in kaart onderaan in "map" symbool).

Ook het verschil in WFS-data en SOS-data is mogelijk niet duidelijk: de 
WFS data is uitgebreider, meer velden/kolommen. De SOS kan alleen enkele 
sensor-waarden geven bijv "value" maar niet de "raw_value", 
min/max/sample_count etc.