smartemission/smartemission #17: Check calibration still ok after upgrades

justb4, 2016-09-24:
Calibration has been updated on the test server to the latest GitHub version. Also the Stetl Docker image has been rebuilt to the latest SciPi (github version, see Docker file: https://github.com/Geonovum/smartemission/blob/master/docker/stetl/Dockerfile). The data conversion for last values shows quite some new warnings and errors though. May be mismatches between (pickled) versions? See e.g. the log output for generated last values for the SmartApp (http://test.smartemission.nl/smartapp/):

```
root@SETEST:~/git/etl# ./last.sh 
2016-09-24 17:26:15,093 util INFO Found cStringIO, good!
2016-09-24 17:26:15,109 util INFO Found lxml.etree, native XML parsing, fabulous!
2016-09-24 17:26:15,149 util INFO Found GDAL/OGR Python bindings, super!!
2016-09-24 17:26:15,157 main INFO Stetl version = 1.1.0rc2
2016-09-24 17:26:15,158 main INFO Found args file at: options/docker.args
2016-09-24 17:26:15,158 ETL INFO INIT - Stetl version is 1.1.0rc2
2016-09-24 17:26:15,158 ETL INFO Config/working dir = /root/git/etl
2016-09-24 17:26:15,158 ETL INFO Reading config_file = last.cfg
2016-09-24 17:26:15,158 ETL INFO Substituting 18 args in config file from args_dict: OrderedDict([('__name__', 'asection'), ('raw_device_url', 'http://whale.citygis.nl/sensors/v1'), ('pg_host', 'postgis'), ('pg_user', 'docker'), ('pg_password', 'docker'), ('pg_port', '5432'), ('pg_database', 'gis'), ('pg_schema', 'smartem'), ('pg_schema_rt', 'smartem_rt'), ('pg_schema_raw', 'smartem_raw'), ('pg_schema_refined', 'smartem_refined'), ('refiner_max_input_records', '4000'), ('publisher_max_input_records', '3000'), ('max_proc_time_secs', '600'), ('api_interval_secs', '1'), ('sos_host', 'sos52n'), ('sos_port', '8080'), ('sos_path', '/sos52n/service')])
2016-09-24 17:26:15,159 ETL INFO Substituting args OK
2016-09-24 17:26:15,159 ETL INFO START
2016-09-24 17:26:15,159 util INFO Timer start: total ETL
2016-09-24 17:26:15,159 chain INFO Assembling Chain: input_raw_sensor_last_api|output_postgres_insert...
/usr/local/lib/python2.7/dist-packages/sklearn/base.py:315: UserWarning: Trying to unpickle estimator StandardScaler from version pre-0.18 when using version 0.19.dev0. This might lead to breaking code or invalid results. Use at your own risk.
  UserWarning)
/usr/local/lib/python2.7/dist-packages/sklearn/base.py:315: UserWarning: Trying to unpickle estimator MLPRegressor from version pre-0.18 when using version 0.19.dev0. This might lead to breaking code or invalid results. Use at your own risk.
  UserWarning)
/usr/local/lib/python2.7/dist-packages/sklearn/base.py:315: UserWarning: Trying to unpickle estimator Pipeline from version pre-0.18 when using version 0.19.dev0. This might lead to breaking code or invalid results. Use at your own risk.
  UserWarning)
2016-09-24 17:26:15,405 input INFO cfg = {'url': 'http://whale.citygis.nl/sensors/v1', 'api_interval_secs': '0', 'class': 'rawsensorapi.RawSensorLastInput', 'sensor_names': 'temperature,humidity,pressure,noiseavg,noiselevelavg,co2,co,no2,o3,coraw,no2raw,o3raw'}
2016-09-24 17:26:15,405 httpinput INFO url=http://whale.citygis.nl/sensors/v1 parameters=None
2016-09-24 17:26:15,406 output INFO cfg = {'input_format': 'record_array', 'database': 'gis', 'replace': 'True', 'host': 'postgis', 'user': 'docker', 'key': 'unique_id', 'table': 'last_device_output', 'password': 'docker', 'class': 'outputs.dboutput.PostgresInsertOutput', 'schema': 'smartem_rt'}
2016-09-24 17:26:15,406 chain INFO Running Chain: input_raw_sensor_last_api|output_postgres_insert
2016-09-24 17:26:15,406 RawSensorAPI INFO Init: fetching device list from URL: "http://whale.citygis.nl/sensors/v1/devices" ...
2016-09-24 17:26:15,429 RawSensorAPI INFO Found   64 devices: [0, 1, 7, 8, 9, 10, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 47, 48, 49, 50, 51, 52, 53, 54, 55, 56, 57, 58, 59, 60, 61, 62, 63, 64, 65, 66, 75662, 828231, 3487232]
2016-09-24 17:26:15,429 dboutput INFO Init: connect to DB
2016-09-24 17:26:15,429 postgis INFO Connecting to dbname=gis user=docker host=postgis port=5432
2016-09-24 17:26:15,438 postgis DEBUG Connected to database gis
2016-09-24 17:26:15,525 RawSensorAPI WARNING device_id=8 sensor=temperature: invalid input for s_temperatureambient: detail=s_temperatureambient not present
2016-09-24 17:26:15,525 RawSensorAPI WARNING device_id=8 sensor=humidity: invalid input for s_humidity: detail=s_humidity not present
2016-09-24 17:26:15,525 RawSensorAPI WARNING device_id=8 sensor=pressure: invalid input for s_barometer: detail=s_barometer not present
2016-09-24 17:26:15,525 RawSensorAPI WARNING device_id=8 sensor=noiseavg: invalid input for ['v_audio0', 'v_audioplus1', 'v_audioplus2', 'v_audioplus3', 'v_audioplus4', 'v_audioplus5', 'v_audioplus6', 'v_audioplus7', 'v_audioplus8', 'v_audioplus9']: detail=v_audio0 not present
2016-09-24 17:26:15,525 RawSensorAPI WARNING device_id=8 sensor=noiselevelavg: invalid input for noiseavg: detail=noiseavg not present
2016-09-24 17:26:15,525 RawSensorAPI WARNING device_id=8 sensor=co2: invalid input for s_co2: detail=s_co2 not present
2016-09-24 17:26:15,525 RawSensorAPI WARNING device_id=8 sensor=co: invalid input for ['s_o3resistance', 's_no2resistance', 's_co2', 's_temperatureambient', 's_temperatureunit', 's_humidity', 's_barometer', 's_lightsensorbottom']: detail=s_o3resistance not present
2016-09-24 17:26:15,525 RawSensorAPI WARNING device_id=8 sensor=no2: invalid input for ['s_o3resistance', 's_no2resistance', 's_co2', 's_temperatureambient', 's_temperatureunit', 's_humidity', 's_barometer', 's_lightsensorbottom']: detail=s_o3resistance not present
2016-09-24 17:26:15,525 RawSensorAPI WARNING device_id=8 sensor=o3: invalid input for ['s_o3resistance', 's_no2resistance', 's_co2', 's_temperatureambient', 's_temperatureunit', 's_humidity', 's_barometer', 's_lightsensorbottom']: detail=s_o3resistance not present
2016-09-24 17:26:15,525 RawSensorAPI WARNING device_id=8 sensor=coraw: invalid input for ['s_coresistance']: detail=s_coresistance not present
2016-09-24 17:26:15,526 RawSensorAPI WARNING device_id=8 sensor=no2raw: invalid input for ['s_no2resistance']: detail=s_no2resistance not present
2016-09-24 17:26:15,526 RawSensorAPI WARNING device_id=8 sensor=o3raw: invalid input for ['s_o3resistance']: detail=s_o3resistance not present
2016-09-24 17:26:15,550 RawSensorAPI WARNING device_id=9 sensor=co2: invalid input for s_co2: detail=s_co2 not present
2016-09-24 17:26:15,550 RawSensorAPI WARNING device_id=9 sensor=co: invalid input for ['s_o3resistance', 's_no2resistance', 's_co2', 's_temperatureambient', 's_temperatureunit', 's_humidity', 's_barometer', 's_lightsensorbottom']: detail=s_o3resistance not present
2016-09-24 17:26:15,550 RawSensorAPI WARNING device_id=9 sensor=no2: invalid input for ['s_o3resistance', 's_no2resistance', 's_co2', 's_temperatureambient', 's_temperatureunit', 's_humidity', 's_barometer', 's_lightsensorbottom']: detail=s_o3resistance not present
2016-09-24 17:26:15,550 RawSensorAPI WARNING device_id=9 sensor=o3: invalid input for ['s_o3resistance', 's_no2resistance', 's_co2', 's_temperatureambient', 's_temperatureunit', 's_humidity', 's_barometer', 's_lightsensorbottom']: detail=s_o3resistance not present
2016-09-24 17:26:15,550 RawSensorAPI WARNING device_id=9 sensor=coraw: invalid input for ['s_coresistance']: detail=s_coresistance not present
2016-09-24 17:26:15,550 RawSensorAPI WARNING device_id=9 sensor=no2raw: invalid input for ['s_no2resistance']: detail=s_no2resistance not present
2016-09-24 17:26:15,550 RawSensorAPI WARNING device_id=9 sensor=o3raw: invalid input for ['s_o3resistance']: detail=s_o3resistance not present
2016-09-24 17:26:15,550 dboutput INFO query is INSERT INTO last_device_output (name,point,value_raw,altitude,value,label,unit,value_stale,time,device_name,unique_id,device_id) VALUES (%s,%s,%s,%s,%s,%s,%s,%s,%s,%s,%s,%s)
2016-09-24 17:26:15,564 dboutput INFO committed 5 records
2016-09-24 17:26:15,587 RawSensorAPI WARNING device_id=10 sensor=noiseavg: invalid input for ['v_audio0', 'v_audioplus1', 'v_audioplus2', 'v_audioplus3', 'v_audioplus4', 'v_audioplus5', 'v_audioplus6', 'v_audioplus7', 'v_audioplus8', 'v_audioplus9']: detail=v_audio0 not present
2016-09-24 17:26:15,587 RawSensorAPI WARNING device_id=10 sensor=noiselevelavg: invalid input for noiseavg: detail=noiseavg not present
[... cut ...] : val(64636000) > max(5000000)
2016-09-24 17:26:17,014 dboutput INFO committed 8 records
2016-09-24 17:26:17,188 RawSensorAPI INFO Processing halted: all devices done
2016-09-24 17:26:17,188 RawSensorAPI INFO Exit
2016-09-24 17:26:17,188 dboutput INFO Exit: disconnect from DB
2016-09-24 17:26:17,189 chain INFO DONE - 65 rounds - chain=input_raw_sensor_last_api|output_postgres_insert 
2016-09-24 17:26:17,189 util INFO Timer end: total ETL time=2.0 sec
2016-09-24 17:26:17,189 ETL INFO ALL DONE
```

pietermarsman, 2016-09-24:
Goeie timing. Ik was net begonnen met modellen leren met de nieuwst versies van scikit-learn. Zal binnenkort modellen met versie 0.19.dev0.

pietermarsman, 2016-09-27:
Is het mogelijk om een vaste versie van sklearn te specificeren? Dat voorkomt dit soort problemen in de toekomst? Dat we bijvoorbeeld op een bepaalde tag gaan zitten in hun git.

justb4, 2016-10-03:
Ja, dat kan in de [Docker file](https://github.com/Geonovum/smartemission/blob/master/docker/stetl/Dockerfile) voor ``Stetl`` die voor alle ETL wordt gebruikt, zie regel 44, daar staat nu: 

```
RUN pip install Cython numpy scipy git+git://github.com/scikit-learn/scikit-learn.git
```

Je kunt de Dockerfile aanpassen of mij aangeven welke ``scikit-learn/`` versie en welk ``pip`` commando gegeven moet worden. Met ``pip`` kan een specifieke versie aangegeven worden.

closed 2016-10-04 by pietermarsman

justb4, 2016-10-10:
OK, de nieuwe Docker image voor Stetl met sklearn 0.18 en laatste GitHub versie SE, draait nu op test-server http://test.smartemission.nl. Wel nog warnings in log, maar die waren er al. Op de productie http://data.smartemission.nl draaien nog vorige versies. Logfiles aangehecht.

Logfile van test:
[smartem_last.log.txt](https://github.com/Geonovum/smartemission/files/519181/smartem_last.log.txt)

Logfile van productie:
[smartem_last_p.log.txt](https://github.com/Geonovum/smartemission/files/519187/smartem_last_p.log.txt)

commit f600692e (2016-10-04): New calibration models based on aws result. Using sklearn 0.18. Fixes #17.