smartemission/smartemission #18: Automatic creation of CSV file of Jose measurements at location 12 and 14

pietermarsman, 2016-10-10:
Columns: s_barometer, s_co2, s_humidity, s_no2resistance, s_o3resistance, s_temperatureambient, s_temperatureunit
Granularity: raw values, 3 times per minute
Maybe seperate files for sensor 12 and 14

justb4, 2016-10-11:
Ok, some questions: 

* columns: with original unconverted sensor values I guess.
* columns: also time in "Zulu" like 2016-06-28T11:00:10.728Z I guess
* all raw values, unvalidated?
* I prefer generating separate file per sensor
* CSVs for what timeframes: one month? one growing CSV from start of collection?

Could be based on current Refiner process, leaving out all steps of pre-validation, conversion and aggregation. For me it would even be easier to write these raw values for these stations to a DB table and generate CSVs from that table. This would even allow collecting CSVs on the fly via HTTP WFS like now the hourly averages e.g.:

http://data.smartemission.nl/geoserver/wfs?request=GetFeature&typename=timeseries_no2&cql_filter=device_id=12&outputformat=csv

``cql_filter`` allows SQL-like queries. 

I that workable: that the learning process collects CSV's via an online HTTP API?

pietermarsman, 2016-10-12:
The training data should be the same as where it is applied on. So the values should be similar to the ones that are used in the sensorconverters.py; the `json_obj`. I guess those values are validated but not converted and not raw, but I am not sure. 

Time is a good thing to have. I definitely want those to match the values to the RIVM values. 

The number of CSV's does not matter. If there are monthly then I will write a for loop to get them all :)

An HTTP API is fine. I'm sure Python can grap CSV over HTTP.

pietermarsman, 2016-12-13:
#6de60a03dd67eec12b738708d3f66ca73dc68f69

pietermarsman, 2016-12-13:
6de60a03dd67eec12b738708d3f66ca73dc68f69

closed 2016-12-23 by pietermarsman