smartemission/smartemission #24: Sensor 71 & 73 geven geen no2 data in de actuele data in SOSEMU

bdg-intemo, 2016-10-28:


justb4, 2016-10-28:
In de log file zie ik een error melding voor deze stations: ``s_temperatureambient`` en ``s_barometer`` zijn 0. Daarom kan de no2 indicator in ug/m3 niet berekend worden ondanks dat ``NO2Raw`` wel beschikbaar is. Dit zie ik vaker bij andere sensors in Nijmegen. Dit is niet gerelateerd aan #25 overigens.

```
2016-10-28 21:42:04,793 dboutput INFO committed 12 records
2016-10-28 21:42:04,821 RawSensorAPI WARNING device_id=71 sensor=co: invalid input for ['s_o3resistance', 's_no2resistance', 's_co2', 's_temperatureambient', 's_temperatureunit', 's_humidity', 's_barometer', 's_lightsensorbottom']: detail=s_temperatureunit: val(0) < min(233150)
2016-10-28 21:42:04,821 RawSensorAPI WARNING device_id=71 sensor=no2: invalid input for ['s_o3resistance', 's_no2resistance', 's_co2', 's_temperatureambient', 's_temperatureunit', 's_humidity', 's_barometer', 's_lightsensorbottom']: detail=s_temperatureunit: val(0) < min(233150)
2016-10-28 21:42:04,821 RawSensorAPI WARNING device_id=71 sensor=o3: invalid input for ['s_o3resistance', 's_no2resistance', 's_co2', 's_temperatureambient', 's_temperatureunit', 's_humidity', 's_barometer', 's_lightsensorbottom']: detail=s_temperatureunit: val(0) < min(233150)
2016-10-28 21:42:04,830 dboutput INFO committed 9 records
2016-10-28 21:42:04,862 dboutput INFO committed 12 records
2016-10-28 21:42:04,882 RawSensorAPI WARNING device_id=73 sensor=temperature: invalid input for s_temperatureambient: detail=s_temperatureambient: val(0) < min(233150)
2016-10-28 21:42:04,882 RawSensorAPI WARNING device_id=73 sensor=pressure: invalid input for s_barometer: detail=s_barometer: val(0) < min(20000)
2016-10-28 21:42:04,882 RawSensorAPI WARNING device_id=73 sensor=co2: invalid input for s_co2: detail=s_co2: val(8480000) > max(5000000)
2016-10-28 21:42:04,882 RawSensorAPI WARNING device_id=73 sensor=co: invalid input for ['s_o3resistance', 's_no2resistance', 's_co2', 's_temperatureambient', 's_temperatureunit', 's_humidity', 's_barometer', 's_lightsensorbottom']: detail=s_co2: val(8480000) > max(5000000)
2016-10-28 21:42:04,882 RawSensorAPI WARNING device_id=73 sensor=no2: invalid input for ['s_o3resistance', 's_no2resistance', 's_co2', 's_temperatureambient', 's_temperatureunit', 's_humidity', 's_barometer', 's_lightsensorbottom']: detail=s_co2: val(8480000) > max(5000000)
2016-10-28 21:42:04,882 RawSensorAPI WARNING device_id=73 sensor=o3: invalid input for ['s_o3resistance', 's_no2resistance', 's_co2', 's_temperatureambient', 's_temperatureunit', 's_humidity', 's_barometer', 's_lightsensorbottom']: detail=s_co2: val(8480000) > max(5000000)
```

justb4, 2016-10-28:
Hier de gehele logfile voor de sosemu (last values ophalen)
[smartem_last.log.txt](https://github.com/Geonovum/smartemission/files/559300/smartem_last.log.txt)
, bijv device 25 geeft al tijden geen temperatuur....

justb4, 2016-10-28:
Hmm mogelijk toch gerelateerd aan #25 na-ijl effect, 
[image]
want even later zie ik in Whale viewer en http://whale.citygis.nl/sensors/v1/devices/71/timeseries/20161028/20 wel temperatuurwaarden voor dezelfde tijdspanne. Zie plaatje.

bdg-intemo, 2016-11-03:
Sensor 73 staat op dit moment ook niet in de smartapp, maar wel online in de Whale viewer. Een uur geleden stond hij geloof ik nog wel in de smartapp.

justb4, 2016-11-03:
```
2016-11-03 11:18:05,450 RawSensorAPI WARNING device_id=73 sensor=temperature: invalid input for s_temperatureambient: detail=s_temperatureambient: val(0) < min(233150)
2016-11-03 11:18:05,450 RawSensorAPI WARNING device_id=73 sensor=pressure: invalid input for s_barometer: detail=s_barometer: val(0) < min(20000)
2016-11-03 11:18:05,451 RawSensorAPI WARNING device_id=73 sensor=co: invalid input for ['s_o3resistance', 's_no2resistance', 's_co2', 's_temperatureambient', 's_temperatureunit', 's_humidity', 's_barometer', 's_lightsensorbottom']: detail=s_temperatureambient: val(0) < min(233150)
2016-11-03 11:18:05,451 RawSensorAPI WARNING device_id=73 sensor=no2: invalid input for ['s_o3resistance', 's_no2resistance', 's_co2', 's_temperatureambient', 's_temperatureunit', 's_humidity', 's_barometer', 's_lightsensorbottom']: detail=s_temperatureambient: val(0) < min(233150)
2016-11-03 11:18:05,451 RawSensorAPI WARNING device_id=73 sensor=o3: invalid input for ['s_o3resistance', 's_no2resistance', 's_co2', 's_temperatureambient', 's_temperatureunit', 's_humidity', 's_barometer', 's_lightsensorbottom']: detail=s_temperatureambient: val(0) < min(233150)
```

Lijkt op wat ik eerder zag, naijlen huidge uurwaarden, bijv  https://github.com/Geonovum/smartemission/issues/24#issuecomment-257012314
en dan
https://github.com/Geonovum/smartemission/issues/24#issuecomment-257015781

justb4, 2016-11-03:
En uit whale api ; ``"s_temperatureambient":0,``

```
{"id":"73","p_unitserialnumber":73,"time":"2016-11-03T11:00:05.8170355Z","s_longitude":5364741,"s_latitude":54616957,"s_satinfo":94985,"v_audioplus9":2566183,"u_audioplus9":2566183,"t_audioplus9":2697512,"s_audioplus9":2631720,"v_audioplus8":2434341,"u_audioplus8":2434341,"t_audioplus8":2697260,"s_audioplus8":2499877,"v_audioplus7":2368808,"u_audioplus7":2368806,"t_audioplus7":3225141,"s_audioplus7":2434602,"v_audioplus6":2630183,"u_audioplus6":2433318,"t_audioplus6":3549747,"s_audioplus6":2695718,"v_audioplus5":3158064,"u_audioplus5":2829868,"t_audioplus5":4013112,"s_audioplus5":3223856,"v_audioplus4":3025449,"u_audioplus4":2762536,"t_audioplus4":3551282,"s_audioplus4":3091240,"v_audioplus3":2632233,"u_audioplus3":2499622,"t_audioplus3":2963269,"s_audioplus3":2697768,"v_audioplus2":2631465,"u_audioplus2":2434340,"t_audioplus2":4074797,"s_audioplus2":2435114,"v_audioplus1":2696998,"u_audioplus1":2368290,"t_audioplus1":2830381,"s_audioplus1":2763048,"v_audio0":2235392,"u_audio0":1971200,"t_audio0":2564608,"s_audio0":2301696,"s_o3resistance":15108,"s_no2resistance":88174,"s_coresistance":325384,"s_rain":0,"s_co2":2388000,"s_bottomswitches":0,"s_rgbcolor":16445951,"s_lightsensorblue":3439,"s_lightsensorgreen":3261,"s_lightsensorred":3379,"s_acceleroz":758,"s_acceleroy":521,"s_accelerox":502,"s_barometer":0,"s_lightsensorbottom":47,"s_lightsensortop":3373,"s_humidity":62284,"s_temperatureambient":0,"s_temperatureunit":300025,"s_secondofday":46682,"s_rtcdate":1093684,"s_rtctime":801283,"p_totaluptime":712494,"p_sessionuptime":710694,"p_basetimer":6,"p_errorstatus":0,"p_11":917504,"p_10":326108,"p_powerstate":3983,"p_18":184549571,"p_17":184549571,"p_coheatermode":184549571}
```

RobertKieboom, 2016-11-09:
Als ik naar de sensorviewer op Whale kijk, dan zie ik de NO2 data:

[image]

In dit geval voor node 71, maar ik zie ook data van de andere twee nodes. Wel is er een gat te zien van ongeveer een halfuur, maar dat heeft volgens mij te maken met de heater, die niet altijd aan is. Als de heater uit is, dan wordt er geen NO2 gemeten,

avdcruyssen, 2016-11-09:
Dat klopt inderdaad. De gassensoren volgens een temperatuurprofiel, wat inhoudt dat ze gedurende het afkoelen en opwarmen geen data doorsturen. De data wordt dan ook in het geheel niet gestuurd. Er is wel een timer variabele die aangeeft in welke stadium de sensor verkeerd (afkoelen, uit, opwarmen), maar dit is een andere identifier dan de gebruikelijke sensordata.

In Nijmegen geldt dit voor CO, NO2, O3 en CO2
In Utrecht geldt dit alleen voor CO, NO2 en O3. CO2 staat, dankzij hernieuwde inzichten tegenwoordig permanent aan, en heeft om een 96 uur een 10 minuten inactiviteit t.b.v. zelfcalibratie.

Zie hieronder de ingestelde data, voor Nijmegen, voor Utrecht moet je de CO2 waarden vergeten.

[image]

justb4, 2016-11-10:
Via #25 is wel een probleem met harvesting opgelost, maar dit "dicht niet alle gaten". Als je boven leest geeft de SOS API (en andere APIs bijv WMS en WFS) twee indicatoren door voor NO2, CO en O3: een ruwe (weerstands-) waarde en een berekende/gecalibreerde (volgens ANN-methode Pieter) waarde in ``ug/m3``. Bijv voor NO2 heten de indicatoren ``no2raw`` (uit ``s_no2resistance ``) en ``no2``. Om ``no2`` te berekenen moeten een serie input ``s_`` waarden aanwezig zijn volgens de [ETL sensor definities](../blob/master/etl/sensordefs.py#L503). Dus aanwezig en in juiste waarde-range. Voor ``no2`` uiteraard ``s_no2resistance`` maar ook meteo en andere gassen (bijv O3 en NO2 beinvloeden elkaar). Met name ``s_temperatureambient`` is bij meerdere stations afwezig, m.n. bij 71 en 73. Dit is ook duidelijk in de log files te zien voor 71 en 73:

```
2016-11-10 14:15:04,312 RawSensorAPI WARNING device_id=71 sensor=co: invalid input for ['s_o3resistance', 's_no2resistance', 's_co2', 's_temperatureambient', 's_temperatureunit', 's_humidity', 's_barometer', 's_lightsensorbottom']: detail=s_temperatureunit: val(0) < min(233150)
2016-11-10 14:15:04,312 RawSensorAPI WARNING device_id=71 sensor=no2: invalid input for ['s_o3resistance', 's_no2resistance', 's_co2', 's_temperatureambient', 's_temperatureunit', 's_humidity', 's_barometer', 's_lightsensorbottom']: detail=s_temperatureunit: val(0) < min(233150)
2016-11-10 14:15:04,312 RawSensorAPI WARNING device_id=71 sensor=o3: invalid input for ['s_o3resistance', 's_no2resistance', 's_co2', 's_temperatureambient', 's_temperatureunit', 's_humidity', 's_barometer', 's_lightsensorbottom']: detail=s_temperatureunit: val(0) < min(233150)
2016-11-10 14:15:04,323 dboutput INFO committed 9 records
2016-11-10 14:15:04,345 RawSensorAPI WARNING id=72-o3  invalid output for o3: detail=o3: val(-0.37335204103) < min(0)
2016-11-10 14:15:04,357 dboutput INFO committed 11 records
2016-11-10 14:15:04,377 RawSensorAPI WARNING device_id=73 sensor=temperature: invalid input for s_temperatureambient: detail=s_temperatureambient: val(0) < min(233150)
2016-11-10 14:15:04,377 RawSensorAPI WARNING device_id=73 sensor=pressure: invalid input for s_barometer: detail=s_barometer: val(0) < min(20000)
2016-11-10 14:15:04,378 RawSensorAPI WARNING device_id=73 sensor=co2: invalid input for s_co2: detail=s_co2: val(9202000) > max(5000000)
2016-11-10 14:15:04,378 RawSensorAPI WARNING device_id=73 sensor=co: invalid input for ['s_o3resistance', 's_no2resistance', 's_co2', 's_temperatureambient', 's_temperatureunit', 's_humidity', 's_barometer', 's_lightsensorbottom']: detail=s_o3resistance: val(1833) < min(3000)
2016-11-10 14:15:04,378 RawSensorAPI WARNING device_id=73 sensor=no2: invalid input for ['s_o3resistance', 's_no2resistance', 's_co2', 's_temperatureambient', 's_temperatureunit', 's_humidity', 's_barometer', 's_lightsensorbottom']: detail=s_o3resistance: val(1833) < min(3000)
2016-11-10 14:15:04,378 RawSensorAPI WARNING device_id=73 sensor=o3: invalid input for ['s_o3resistance', 's_no2resistance', 's_co2', 's_temperatureambient', 's_temperatureunit', 's_humidity', 's_barometer', 's_lightsensorbottom']: detail=s_o3resistance: val(1833) < min(3000)
2016-11-10 14:15:04,378 RawSensorAPI WARNING device_id=73 sensor=o3raw: invalid input for ['s_o3resistance']: detail=s_o3resistance: val(1833) < min(3000)
```

Voor 71 is bijv ``s_temperatureunit: val(0) < min(233150)`` dwz 0 graden Kelvin dus invalide. Voor 73 is het ``s_o3resistance: val(1833) < min(3000)``, dus ``s_o3resistance`` buiten range. 

Dat laatste heb ik uit de specs van Antoine, maar mogelijk kan iemand nog een keer goed naar de  [ETL sensor definities](../blob/master/etl/sensordefs.py) kijken?

avdcruyssen, 2016-11-11:
Bij 71 is de unittemperature idd 0, hoe dat kan weet ik ook niet, maar die waarde is niet interessant, de ambient temperature daarentegen is degene die je wilt hebben denk ik. Unit temperature is de temperatuursensor IN de unit, en dient meer voor debug doeleinden.

Bij Ozon is de waarde niet hard, d.w.z. die is gespecificeerd tot 3000, die kan echter wel degelijk verder droppen, maar valt dan buiten de te meten range. Waarschijnlijk is de lucht daar dan goed schoon, en ligt het ruisniveau van de sensor daar lager dan elders.

justb4, 2016-11-11:
Ik hecht nogmaals de volledige logfile aan voor de laatste waarden van alle sensoren, de omzetting/calibratie is dezelfde als voor historie. Te zien is dat het ontbreken van ``s_temperatureambient`` vrij vaak voorkomt en de oorzaak is dat veel gasberekeningen niet kunnen worden gedaan (en dus gaten in historie veroorzaken). ``s_temperatureunit`` veel minder, maar de omzetting (van Pieter) heeft beiden nodig. Mogelijk moeten we anders een recente waarde nemen, bijv tot max 3 uur terug als een indicator ontbreekt. Behalve geluid veranderen meteo en in mindere mate gaswaarden niet heel snel.
 
Ozon minimum (``s_o3resistance``) heb ik verlaagd in Sensor defs.

[smartem_last.log.txt](https://github.com/Geonovum/smartemission/files/585694/smartem_last.log.txt)

avdcruyssen, 2016-11-11:
Zoals ik het begrepen heb, gaan we binnenkort een (fysiek) rondje doen om e.e.a. weer terug in de lucht te krijgen. We hebben een probleem ontdekt dat zich manifesteert na een bepaalde tijd, en daarvoor een oplossing bedacht, echter deze moet nog wel geimplementeerd worden. We moeten echter even tijd vinden, en de betreffende sensoren opsnorren.

pietermarsman, 2017-04-04:
Misschien dit issue sluiten omdat het probleem aan de hardware kant ligt en niet binnen deze repo opgelost kan worden? 

Ik heb een nieuw issue aangemaakt voor het robuust zijn van tijdelijk ontbrekende variabelen: https://github.com/Geonovum/smartemission/issues/52

justb4, 2017-04-05:
Ok.

closed 2017-04-05 by justb4