smartemission/smartemission #25: Er zitten "gaten" in de historische data van de 52 North API

bdg-intemo, 2016-10-28:


justb4, 2016-10-28:
Ben aan het onderzoeken. Er is op sommige momenten iets vreemds met de data die ik uit de Whale-server krijg:  het lijkt erop dat deze soms achterloopt voor de gassen co, o3 en no2, nooit bijv voor co2 en andere indicatoren als meteo en geluid. Maar dan later weer "bijgewerkt" wordt, een soort "naijlen" effect. Echter het "uur" is dan al voorbij en de data uitgelezen en naar SOS gepubliceerd. 

Bijv. http://whale.citygis.nl/sensors/v1/devices/70/timeseries/20161028/11 voor sensor 70, 28 okt tussen 10 en 11 uur UTC, 12 en 13 uur onze tijd. Tot 10:15 UTC zijn er geen no2-metingen.

Nogmaals: ik ben aan het onderzoeken, het kan een mismatch zijn, want op zich verwerkt het data-platform het gehele uur voor gemiddelden berekening. Het kan zijn dat de Whale-server op sommige momenten druk is en daardoor achter loopt. Maar het blijft vreemd waarom dan alleen de 3 gassen niet doorkomen....

justb4, 2016-10-28:
[image]

Dit plaatje verduidelijkt hopelijk e.e.a. Het laat de geaggregeerde uurwaarden voor ``no2_resistance`` (``no2raw``) van sensor 70 op 27 en 28 okt 2016 zien.  De kolom ``insert_time`` is in NL time (momenteel UTC+2) het moment waarop het record wordt insert vanuit de cronjob. De kolommen ``day`` en ``hour`` zijn de Whale-data UTC dag/tijd. Bijv uur 0 op 27 okt is tussen 23:00 en 24:00 UTC op 26 okt (tussen 0100 en 0200 UTC+2). De aggregatie wordt steeds na het voorbije uur gedaan.  Dus voor uur 0 UTC om 02:35 UTC+2. Goed te zien is dat hoe dichter het aggregatie-tijdstip na het hele uur, bijv 5 minuten na hele uur, is, des te lager de ``sample_count`` is. Na uur 0 is er al een gat van 3 uur en een ``sample_count`` van 6.  Het patroon is duidelijk: hoe later de ``insert_time`` binnen het uur na ``hour``, des te hoger de ``sample_count``. De gaten zijn steeds rond een ``sample_count`` < 10.

Deze job liep eerst om de 5 minuten op de ruwe harvested data die om de 15 minuten wordt binnengehaald en opgeslagen in ons DB data-platform. Heb deze cronjob op 28 okt plm 21:00 aangepast om alleen te lopen op minuten 30,35,40,45,50,55. Ik geef de Whale server a.h.w. de tijd om "na te ijlen" van het voorgaande uur. Goed is te zien dat de laatste 3 records dan 180 waarden (``sample_count`` = 180) middelen, i.e. 3 metingen per minuut, zoals verwacht.

justb4, 2016-10-30:
Probleem gevonden en denk opgelost. Nu op test server testend...

Probleem was tweeledig:
*  in sommige gevallen werd er "te vroeg geoogst", i.e. tijdens lopende uur, door data-platform. Zie fix d7a3a1312904b315bb869030fcf182f30abcbf1e
*  Daarbij kwam het na-ijl effect: tijdens lopende uur waren nog niet alle waarden aanwezig, ook soms niet voor temperatuur. 

Dus deze combinatie zorgde ervoor dat er voor de gassen "gaten" waren, maar niet bijv voor geluid.

Het is alleen wel zo dat de hele historie opnieuw opgebouwd moet worden. Dat moest sowieso omdat ergens in juli-sep ook de calibratie verbeterd is. Zal aantal dagen duren. Ik doe dat op de test-server en zet dan historie in een keer over.

justb4, 2016-10-31:
Vanaf zondag 30 okt 2016 17:00+1 moeten er geen gaten meer in de data voorkomen anders dan dat een sensor zelf is uitgevallen of de Ahale server API anderzijds geen metingen heeft. Laat weten of dit toch nog gebeurt.  De historie daarvoor heeft dus nu nog wel gaten. Die worden gedicht zodra de historie op de testserver opnieuw is opgebouwd en overgezet naar productie server. Dat moest sowieso gebeuren omdat Pieter calibratie heeft ingevoerd en in sept 2016 nog verbeterd.

bdg-intemo, 2016-10-31:
Bedankt voor het nakijken, ik ga het in de gaten houden!

Please consider the environment before printing this email

This e-mail, and any files transmitted with it, is confidential and is intended solely for the use of the person or entity to whom it is addressed. If you have received this e-mail in error, please notify the sender and delete it from your records.

On all our offers, sales and deliveries our General Terms of Delivery are applicable, excluding all other Terms and Conditions. Our terms have been deposited at the Chamber of commerce Eindhoven (The Netherlands) august 23rd, 2001, reference number 17065041, a copy of these terms is enclosed with all quotations, a hardcopy also can be send on your request. You can find our General Terms online at http://www.intemo.com/en/about-intemo/terms-and-conditions-delivery/
Op al onze aanbiedingen, verkopen en leveringen zijn, met uitsluiting van alle andere voorwaarden, onze algemene voorwaarden van toepassing. Deze zijn gedeponeerd bij de Kamer van Koophandel te Eindhoven onder nummer 17065041 per 23 augustus 2001. Bij iedere offerte hoe ook verzonden zijn de algemene voorwaarden bijgevoegd, een kopie kan op aanvraag ook per post worden toegezonden. U kunt onze algemene voorwaarden online inzien op http://www.intemo.com/nl/over-intemo/voorwaarden/

bdg-intemo, 2016-11-03:
De sensoren zijn gisteren nagekeken, maar op de Smartapp zie ik nog steeds geen gecalibreerde waarden verschijnen.

Enig idee waardoor dat kan komen? Of draaien de fixes alleen nog op de testserver?

justb4, 2016-11-03:
De meeste fixes draaien ook op de productie server, m.n. om nooit laatste uur dat bezig is op te halen van Whale server. Ik zie in Utrecht nu redelijk historie. De testserver is historie opnieuw aan het genereren, duurt plm 5 dagen. Heeft vnl te maken dat ik de Whale server niet wil "overvoeren".

Kun je aangeven om welke sensoren het gaat? Gaten in historie van gassen CO, NO2 en O3 kunnen verschillende oorzaken hebben. Wat ik meeste zie is dat meteo ontbreekt waardoor de calibratie niet kan worden uitgevoerd. Hieronder de log van "last values" zoals door SmartApp gebruikt en gehele log aangehecht. Bij Sensor 16:

```
2016-11-03 11:18:03,222 RawSensorAPI WARNING device_id=16 sensor=temperature: invalid input for s_temperatureambient: detail=s_temperatureambient: val(0) < min(233150)
2016-11-03 11:18:03,223 RawSensorAPI WARNING device_id=16 sensor=co: invalid input for ['s_o3resistance', 's_no2resistance', 's_co2', 's_temperatureambient', 's_temperatureunit', 's_humidity', 's_barometer', 's_lightsensorbottom']: detail=s_temperatureambient: val(0) < min(233150)
2016-11-03 11:18:03,223 RawSensorAPI WARNING device_id=16 sensor=no2: invalid input for ['s_o3resistance', 's_no2resistance', 's_co2', 's_temperatureambient', 's_temperatureunit', 's_humidity', 's_barometer', 's_lightsensorbottom']: detail=s_temperatureambient: val(0) < min(233150)
2016-11-03 11:18:03,223 RawSensorAPI WARNING device_id=16 sensor=o3: invalid input for ['s_o3resistance', 's_no2resistance', 's_co2', 's_temperatureambient', 's_temperatureunit', 's_humidity', 's_barometer', 's_lightsensorbottom']: detail=s_temperatureambient: val(0) < min(233150)
```
Er is geen temperatuur dus de 3 gassen kunnen niet berekend worden uit "raw", bijv ``no2raw`` weerstands-waarden.

[last-values-161103-1118.txt](https://github.com/Geonovum/smartemission/files/568668/last-values-161103-1118.txt)

bdg-intemo, 2016-11-03:
Oké, dat lijkt me duidelijk. Ik begreep dat de sensoren gisteren goed functioneerden. Op de smartapp zie ik ook gewoon meteo staan.

Hij hoort eigenlijk ook een beetje bij #24 
Dus bijvoorbeeld sensor 71 en 73.

justb4, 2017-10-03:
Opgelost

closed 2017-10-03 by justb4