smartemission/smartemission #27: Add (un)certainty measurement to reported values

pietermarsman, 2016-11-09:
For example RMSE: 
O3: +- 7.4 ug/m3
NO2: +- 12.6 ug/m3
CO: +- 124.5 ug/m3

pietermarsman, 2017-04-04:
Nog beter: mean absolute error (beter te interpreteren)

pietermarsman, 2017-04-11:
De smartapp visualiseert de sos api. Zit daarin de mogelijkheid om de onzekerheid mee te geven? Als dat zo is kan aan elke meting de onzekerheid toegevoegd worden (van het huidige model) en dan kan die getoond worden. 

Anders word het lastiger. Dan moet de smartapp rechtstreeks de postgres aanspreken. Zijn er nog andere opties?

justb4, 2017-04-11:
IK ben bezig met inregelen SensorThings API (STA) als Docker (Geodan "GOST" Open Source product https://github.com/Geodan/gost). Doel is om SOS te vervangen door STA, en SmartApp op STA  aan te sluiten. 
Er wordt nu op test gepubliceerd (vanuit Refined DB Data PG) naar deze STA. Zie bijv 
http://test.smartemission.nl/gost/v1.0. Er zijn nog wat oneffenheden, ben ik met Geodan aan bezig. 

STA kent bijv voor Observations een `resultQuality`, daar kunnen we iets mee doen. ZIe bijv documentatie STA http://developers.sensorup.com/docs/ (onder Observations), maar er zijn meer mogelijkheden, "vrije properties" met STA.

pietermarsman, 2017-06-06:
Voorlopig ligt dit dus niet in de planning. Mogelijk later heropenen als de SmartApp STA visualiseert ipv SOS.

justb4, 2017-06-07:
Mag van mij open blijven: temeer reden om op STA ipv SOS te standaardiseren :-).