smartemission/smartemission #36: Replace sklearn MLPRegressor by a Keras recurrent neural net

pietermarsman, 2017-01-15:
- Eliminates the need for manual filtering
- Better prediction performance

justb4, 2018-12-03:
@pietermarsman I would like to investigate ANN Calibration for AirSensEUR sensor stations attached to SE platform: https://data.smartemission.nl/grafana-dc/d/HVSBmbHmz/airsenseur-netherlands-deploy we have raw data in mV. The 5 stations were placed at RIVM station Breukelen for 2 months: https://justobjects.nl/emit-5-assembling-and-deploying-5-airsenseurs/

Trying to enter this topic (of ANN):

* is Keras a better choice than Scikit-learn ?
* if yes, what would be the optimal Keras back-end: TensorFlow or Theano or...?
* any other insights?

pietermarsman, 2018-12-24:
Hi @justb4! The choice of packages depends on what you want to do. Keras is much more flexible than  sklearn, it has more optimisers, more types of layers and more activation functions. Scikit-learn on the other hand provides several machine-learning models (besides) ANN with the same api. If you are going to stick with ANN's, I would suggest Keras. 
Tensorflow is a package developed by Google and well maintained. Theano is developped by a university team and they don't maintain it anymore since end of 2017. It still works properly though. I would advice to use TensorFlow. 

Good read: https://developers.google.com/machine-learning/guides/rules-of-ml/

justb4, 2019-01-07:
Ok thanks @pietermarsman though we need a project/funding to actually work on this, as it is not a small change. Things are in the works...