smartemission/smartemission #38: Check if RIVM SOS uses well-defined time

pietermarsman, 2017-01-16:
Mail Just:

RIVM is in verleden wel eens "slordig" met tijd geweest: welk tijdzone
aangehouden werd, ik dacht NL tijd en zomertijd/wintertijd zelfs. Daarom
hoop ik wel dat hun SOS de juiste tijd geeft. Het lijkt soms of ze in
trends achterlopen (SE voor), kan met uren mismatch te maken hebben.
Ik weet bijna 100% zeker dat SE wel goed met tijd omgaat: alles als UTC
opgeslagen, maar je weet maar nooit.

justb4, 2017-01-16:
Een manier om te vergelijken: op LML site staan RIVM LML (uur)waarden voor laatste maand:
http://www.lml.rivm.nl/sos/ bijv NO2 http://www.lml.rivm.nl/sos/2017011321-NO2.xml (13 jan 20:00-21:00 maar welke tijdzone?):

```
<ROW>
<OPST_OPDR_ORGA_CODE>RIVM</OPST_OPDR_ORGA_CODE>
<STAT_NUMMER>741</STAT_NUMMER>
<STAT_NAAM>Nijmegen-Graafseweg</STAT_NAAM>
<MCLA_CODE>stad verkeer</MCLA_CODE>
<MWAA_WAARDE>54.94</MWAA_WAARDE>
<MWAA_BEGINDATUMTIJD>20170113200000</MWAA_BEGINDATUMTIJD>
<MWAA_EINDDATUMTIJD>20170113210000</MWAA_EINDDATUMTIJD>
</ROW>
<ROW>
<OPST_OPDR_ORGA_CODE>RIVM</OPST_OPDR_ORGA_CODE>
<STAT_NUMMER>742</STAT_NUMMER>
<STAT_NAAM>Nijmegen-Ruyterstraat</STAT_NAAM>
<MCLA_CODE>stad achtergr</MCLA_CODE>
<MWAA_WAARDE>39.57</MWAA_WAARDE>
<MWAA_BEGINDATUMTIJD>20170113200000</MWAA_BEGINDATUMTIJD>
<MWAA_EINDDATUMTIJD>20170113210000</MWAA_EINDDATUMTIJD>
</ROW>
```

En dit vergelijken met SOS via bijv http://inspire.rivm.nl/sos/eaq/api/v1/ en wat SmartEmission in InfluxDB (``rivm`` tabel) heeft en bijv via InfluxDB Query of Grafana Dashboard te zien is: 
http://test.smartemission.nl/grafana/dashboard/db/firstrivm

[image]

justb4, 2017-01-16:
Alles lijkt er wel op dat althans SE de tijden uit de RIVM SOS als UTC beschouwt bijv directe query uit InfluxDB:

```
Uit query InlfuxDB:
select * from rivm where "component" = 'nitrogen_dioxide__air_' AND "station" = 'nijmegen_graafseweg' 

["2017-01-13T15:00:00Z","nitrogen_dioxide__air_","u1hnvm3","nijmegen_graafseweg",41.09],
["2017-01-13T16:00:00Z","nitrogen_dioxide__air_","u1hnvm3","nijmegen_graafseweg",43.58],
["2017-01-13T17:00:00Z","nitrogen_dioxide__air_","u1hnvm3","nijmegen_graafseweg",32.79],
["2017-01-13T18:00:00Z","nitrogen_dioxide__air_","u1hnvm3","nijmegen_graafseweg",35.54],
["2017-01-13T19:00:00Z","nitrogen_dioxide__air_","u1hnvm3","nijmegen_graafseweg",39.97],
["2017-01-13T20:00:00Z","nitrogen_dioxide__air_","u1hnvm3","nijmegen_graafseweg",49.34],
["2017-01-13T21:00:00Z","nitrogen_dioxide__air_","u1hnvm3","nijmegen_graafseweg",54.94],
["2017-01-13T22:00:00Z","nitrogen_dioxide__air_","u1hnvm3","nijmegen_graafseweg",32.73],
["2017-01-13T23:00:00Z","nitrogen_dioxide__air_","u1hnvm3","nijmegen_graafseweg",16.56],
["2017-01-14T00:00:00Z","nitrogen_dioxide__air_","u1hnvm3","nijmegen_graafseweg",12.59],
["2017-01-14T01:00:00Z","nitrogen_dioxide__air_","u1hnvm3","nijmegen_graafseweg",14.11]
```

Om "2017-01-13T21:00:00Z" is waarde ``54.94``. Grafana hanteerde weer browsertijd, maar als we die op UTC zetten, is ``54.94`` ook om 21:00:

[image]
 
Hmm dan moeten we ervan uitgaan dat de tijden uit de LML XML ook UTC zijn, ik dacht altijd dat dit lokale tijden waren. Of we maken zelf ergens een fout in de keten: ``SOSInput-->INfluxDBOutput``, daar gebeuren aantal datetime manipulaties:

Doorgaand op de timestamp uit SOS 1484337600(000) zie ik in 
https://github.com/Geonovum/smartemission/blob/master/etl/sosinput.py#L119

```
elem['datetime'] = datetime.fromtimestamp(elem['timestamp'] / 1000)
```

Ik denk dat ``datetime.utcfromtimestamp()`` moet gebruikt worden, anders wordt lokale datetime genomen en wordt bij conversie terug naar timestamp ten onrechte ``21:00`` als UTC beschouwd in de ``InfluxDBOutput``, zie https://github.com/Geonovum/smartemission/blob/master/etl/influxdboutput.py#L127 en https://github.com/Geonovum/smartemission/blob/master/etl/influxdboutput.py#L191 bijv:
 
```
nusa:foto just$ python
Python 2.7.12 (default, Jun 29 2016, 14:05:02) 
[GCC 4.2.1 Compatible Apple LLVM 7.3.0 (clang-703.0.31)] on darwin
Type "help", "copyright", "credits" or "license" for more information.

>>> from datetime import datetime
>>> t=datetime.fromtimestamp(1484337600)
>>> print str(t)
2017-01-13 21:00:00
>>> t=datetime.utcfromtimestamp(1484337600)
>>> print str(t)
2017-01-13 20:00:00
```

Daarom denk ik dat de oorspronkelijke ``20:00`` UTC tijd verschoven is naar ``21:00`` UTC in bijv Grafana (en dus 22:00 in lokale browser-tijd). Het is nogal complex. Ik probeer daarom intern en in DB uitsluitend UTC gebruiken, vandaar de conversies. Hopelijk klopt mijn redenatie en hoeft alleen ``SOSInput`` aangepast.

justb4, 2017-01-17:
Bevestiging door RIVM:

> de tijd is de tijd die in het LML wordt gehanteerd en dat is GMT+1, zonder DST
> Derko

Dus onze aannames kloppen en de RIVM SOS geeft dus correcte Posix system timestamps af.

justb4, 2017-01-17:
Mail Just:

> De tijd uit de LML XML is altijd Zulu/GMT/UTC+1, zonder zomer/wintertijd.
> Dan klopt e.e.a. precies en dus ook jouw commit want anders is de data in InfluxDB 1 uur, soms 2 uur, later bijv
> 
> select * from rivm where "component" = 'nitrogen_dioxide__air_' AND "station" = 'nijmegen_graafseweg' 
> 
> ["2017-01-13T21:00:00Z","nitrogen_dioxide__air_","u1hnvm3","nijmegen_graafseweg",54.94],
> 
> moet worden na jouw fix https://github.com/Geonovum/smartemission/commit/108cb61f98989b344152dfdbbfe56a14577cfa7f :
> 
> ["2017-01-13T20:00:00Z","nitrogen_dioxide__air_","u1hnvm3","nijmegen_graafseweg",54.94],
> 
> RIVM SOS geeft namelijk overeenkomstige timestamp 1484337600000, dat is UTC 2017-01-13T20:00, gemiddelde NO2 tussen 19:00 en 20:00 UTC (en 20:00-21:00 UTC+1).

pietermarsman, 2017-04-04:
Dus als ik het goed begrijp is nu alles kloppend in UTC? Dan kan dit issue worden gesloten.

justb4, 2017-04-05:
Ik zou het nog 1 keer na moeten gaan nu we zomertijd hebben...

justb4, 2017-10-03:
Ja aanname blijft overeind, dus meenemen in issue #58 : 

* LML waarden in XML altijd in GMT+1
* waarden uit SOS waar we uit harvesten in UTC
* waarden in SOS viewer lokale tijd

Zie plaatje: bijv de NO2 waarde Nijmegen-Ruyterstraat op 3 okt 2017, om *17:00 GMT+2* (onze tijd) is `11.78` (de gemiddelde NO2-waarde tussen 16:00 en 17:00 GMT+2). Dus:

* in LML XML (`MWAA_EINDTIJD`) *1600* (namelijk GMT+1)
* in SOS JSON UTC timestamp *1507042800000* geeft *15:00 GMT* en dus *17:00 GMT+2*

[image]

closed 2017-10-03 by justb4

justb4, 2018-01-05:
Nogmaals check nu voor wintertijd (in NL nu 1 uur voor op UTC/GMT). Klopt ook. Zie plaatje: om 15:00 GMT+1 is NO2 in Ruyterstraat 34.96 ug/m3 (35 afgerond). Klopt met alles, ook bijv UTC timestamp 1515074400000 uit hun SOS, en via InfluxDB in SE Grafana Viewer.

[image]