smartemission/smartemission #39: Calibratie toevoegen aan laatste waarden

pietermarsman, 2017-01-25:
Just: 

> de "last.sh, last.cfg" haalt live waarden op van WHale API (tbv
> SmartApp) op en roept direct de sensorconverters.py aan via
> sensordefs.py. Dus heeft niet de calibrator initialisatie en geeft
> foutmeldingen hierover.
>

pietermarsman, 2017-01-25:
Extra probleem hierbij: de data wordt gefilterd in sensorconverters.py. Hier kan maar 1 proces gebruik van maken want er worden globale variabelen gebruikt.

justb4, 2017-01-25:
Ja klopt, is historisch, ``last`` was er als eerste om wat te laten zien in SmartApp. Ik denk dat we ``sensorconverters.py`` het beste kunnen 'modulariseren' zodat deze o.a. zelf de modellen uit de DB ophaalt. Mogelijk kunnen de globale vars dan in class-instances. Dan kan ``sensorconverters.py`` ook los van Stetl aangeroepen worden.  Ik denk trouwens dat deze globals (en model-instantie) per Stel-proces uniek zijn (niet voor elkaar zichtbaar). Mogelijk ontstaan er ook geheugen problemen bij meerdere model-instanties(?). Maar laten we eerst de andere Stetl processen (alles behalve ``last``) op orde krijgen en dan 'modulariseren'.

commit 41bc1e13 (2017-01-25): closes #39, loads calibration models for last.sh using a CalibrationModelInput
closed 2017-01-25 by pietermarsman