smartemission/smartemission #40: Memory full bij leren calibratie model

pietermarsman, 2017-01-25:
Just: 

> Geheugen probleem "OSError: [Errno 12] Cannot allocate memory"
> calibrator, maar had je aangegegeven:

Error: 

`2017-01-22 11:51:12,537 InfluxDbInput INFO Querying database: SELECT
time, component, geohash, value FROM joseextract WHERE
(geohash='u1hnvkb' or geohash='u1hnvkc' or geohash='u1hnvm3' or
geohash='u178hw8' or geohash = 'u178kt7') AND time >
'2017-01-22T09:59:59Z' LIMIT 1000000
2017-01-22 11:51:12,539 urllib3.connectionpool INFO Resetting dropped
connection: influxdb
2017-01-22 11:51:12,541 InfluxDbInput INFO Received 0 results
2017-01-22 11:51:13,693 Calibration INFO Received rivm data with shape
(45844, 5)
2017-01-22 11:51:13,693 Calibration INFO Received jose data with shape
(3422682, 10)
2017-01-22 11:51:13,693 Calibration INFO Pre-processing geohash and time
2017-01-22 11:51:19,512 Calibration INFO Interpolating RIVM values
towards jose measurements
2017-01-22 11:52:13,411 Calibration INFO RIVM and Jose are merged, new
shape (5471651, 13)
2017-01-22 11:52:13,412 Calibration INFO Filling and dropping missing values
2017-01-22 11:52:19,353 Calibration INFO Missing values:
{u's_no2resistance': 881144, u's_lightsensortop': 0,
u'carbon_monoxide__air_': 0, u'nitrogen_dioxide__air_': 0,
u's_temperatureunit': 626608, u's_barometer': 626608, u's_o3resistance':
881552, 'time': 0, 'geohash': 0, u'ozone__air_': 0,
u's_temperatureambient': 734759, u's_coresistance': 1488382,
u's_humidity': 29}
2017-01-22 11:52:22,220 Calibration INFO Missing values are removed, new
shape (3922930, 13).
2017-01-22 11:58:50,900 Calibration INFO Returning packet of length 3922930
2017-01-22 11:58:52,254 Calibration INFO Receiving packet of size 3922930
2017-01-22 11:59:49,703 Calibration INFO Created data frame with shape
(3922930, 11)
2017-01-22 12:00:03,091 Calibration INFO Sample data frame, keeping 1
out of every 50 rows. New shape (78459, 9)
2017-01-22 12:00:03,160 Calibration INFO Starting randomized cross
validated search to find best parameters. Running 50 iterations with 10
cross validations of -2 cores. Finding relation from s_barometer,
s_coresistance, s_humidity, s_lightsensortop, s_no2resistance,
s_o3resistance, s_temperatureambient, s_temperatureunit to
carbon_monoxide__air_.
2017-01-22 12:00:12,505 dboutput INFO Exit: disconnect from DB
Traceback (most recent call last):
   File "/usr/local/bin/stetl", line 5, in <module>
     pkg_resources.run_script('Stetl==1.1.0rc4', 'stetl')
   File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 528,
in run_script
     self.require(requires)[0].run_script(script_name, ns)
   File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 1394,
in run_script
     execfile(script_filename, namespace, namespace)
   File
"/usr/local/lib/python2.7/dist-packages/Stetl-1.1.0rc4-py2.7.egg/EGG-INFO/scripts/stetl",
line 40, in <module>
     main()
   File
"/usr/local/lib/python2.7/dist-packages/Stetl-1.1.0rc4-py2.7.egg/EGG-INFO/scripts/stetl",
line 31, in main
     etl.run()
   File
"/usr/local/lib/python2.7/dist-packages/Stetl-1.1.0rc4-py2.7.egg/stetl/etl.py",
line 101, in run
     chain.run()
   File
"/usr/local/lib/python2.7/dist-packages/Stetl-1.1.0rc4-py2.7.egg/stetl/chain.py",
line 208, in run
     packet = self.first_comp.process(packet)
   File
"/usr/local/lib/python2.7/dist-packages/Stetl-1.1.0rc4-py2.7.egg/stetl/component.py",
line 185, in process
     packet = self.next.process(packet)
   File
"/usr/local/lib/python2.7/dist-packages/Stetl-1.1.0rc4-py2.7.egg/stetl/component.py",
line 185, in process
     packet = self.next.process(packet)
   File
"/usr/local/lib/python2.7/dist-packages/Stetl-1.1.0rc4-py2.7.egg/stetl/component.py",
line 185, in process
     packet = self.next.process(packet)
   File
"/usr/local/lib/python2.7/dist-packages/Stetl-1.1.0rc4-py2.7.egg/stetl/component.py",
line 174, in process
     packet = self.invoke(packet)
   File "/root/git/etl/calibration.py", line 217, in invoke
     gs.fit(x, y)
   File
"/usr/local/lib/python2.7/dist-packages/sklearn/model_selection/_search.py",
line 1185, in fit
     return self._fit(X, y, groups, sampled_params)
   File
"/usr/local/lib/python2.7/dist-packages/sklearn/model_selection/_search.py",
line 562, in _fit
     for parameters in parameter_iterable
   File
"/usr/local/lib/python2.7/dist-packages/sklearn/externals/joblib/parallel.py",
line 728, in __call__
     n_jobs = self._initialize_backend()
   File
"/usr/local/lib/python2.7/dist-packages/sklearn/externals/joblib/parallel.py",
line 540, in _initialize_backend
     **self._backend_args)
   File
"/usr/local/lib/python2.7/dist-packages/sklearn/externals/joblib/_parallel_backends.py",
line 307, in configure
     self._pool = MemmapingPool(n_jobs, **backend_args)
   File
"/usr/local/lib/python2.7/dist-packages/sklearn/externals/joblib/pool.py",
line 600, in __init__
     super(MemmapingPool, self).__init__(**poolargs)
   File
"/usr/local/lib/python2.7/dist-packages/sklearn/externals/joblib/pool.py",
line 420, in __init__
     super(PicklingPool, self).__init__(**poolargs)
   File "/usr/lib/python2.7/multiprocessing/pool.py", line 159, in __init__
     self._repopulate_pool()
   File "/usr/lib/python2.7/multiprocessing/pool.py", line 223, in
_repopulate_pool
     w.start()
   File "/usr/lib/python2.7/multiprocessing/process.py", line 130, in start
     self._popen = Popen(self)
   File "/usr/lib/python2.7/multiprocessing/forking.py", line 121, in
__init__
     self.pid = os.fork()
OSError: [Errno 12] Cannot allocate memory`

pietermarsman, 2017-01-25:
Gaat fout bij start calibratie, als de data over meerdere threads verspreid moet worden. 

Verbeteringen:
- Geen filtering meer doen, dan kan er meteen worden geresampled
- Enkele core gebruiken voor RandomSearchCV
- 1 op 100 rijen gebruiken

Voor debugging: 
- Memory loggen

pietermarsman, 2017-01-25:
f1e55b449cfdb8e1de55e44b389555432b666512

justb4, 2017-01-25:
De calibrator (met laatste commits) geeft nu andere error, zie
http://test.smartemission.nl/adm/logs/etl/calibrator.log, mogelijk missing typecast?

```
2017-01-25 16:48:11,473 Calibration INFO Received rivm data with shape (46152, 5)
2017-01-25 16:48:11,473 Calibration INFO Received jose data with shape (3519893, 10)
2017-01-25 16:48:11,473 Calibration INFO Pre-processing geohash and time
2017-01-25 16:48:17,499 Calibration INFO Interpolating RIVM values towards jose measurements
2017-01-25 16:49:13,872 Calibration INFO RIVM and Jose are merged, new shape (5628532, 13)
2017-01-25 16:49:13,872 Calibration INFO Filling and dropping missing values
2017-01-25 16:49:19,862 Calibration INFO Missing values: {u's_no2resistance': 896634, u's_lightsensortop': 0, u'carbon_monoxide__air_': 0, u'nitrogen_dioxide__air_': 0, u's_temperatureunit': 626608, u's_barometer': 626608, u's_o3resistance': 897042, 'time': 0, 'geohash': 0, u'ozone__air_': 0, u's_temperatureambient': 734759, u's_coresistance': 1503872, u's_humidity': 29}
2017-01-25 16:49:22,809 Calibration INFO Missing values are removed, new shape (4064321, 13).
2017-01-25 16:49:22,809 Calibration INFO Returning dict with pandas DataFrame as only element
2017-01-25 16:49:22,838 Calibration INFO Receiving packet of size 1
2017-01-25 16:49:23,860 Calibration INFO Created data frame with shape (4064321, 11)
2017-01-25 16:49:23,869 dboutput INFO Exit: disconnect from DB
Traceback (most recent call last):
  File "/usr/local/bin/stetl", line 5, in <module>
    pkg_resources.run_script('Stetl==1.1.0rc4', 'stetl')
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 528, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 1394, in run_script
    execfile(script_filename, namespace, namespace)
  File "/usr/local/lib/python2.7/dist-packages/Stetl-1.1.0rc4-py2.7.egg/EGG-INFO/scripts/stetl", line 40, in <module>
    main()
  File "/usr/local/lib/python2.7/dist-packages/Stetl-1.1.0rc4-py2.7.egg/EGG-INFO/scripts/stetl", line 31, in main
    etl.run()
  File "/usr/local/lib/python2.7/dist-packages/Stetl-1.1.0rc4-py2.7.egg/stetl/etl.py", line 101, in run
    chain.run()
  File "/usr/local/lib/python2.7/dist-packages/Stetl-1.1.0rc4-py2.7.egg/stetl/chain.py", line 208, in run
    packet = self.first_comp.process(packet)
  File "/usr/local/lib/python2.7/dist-packages/Stetl-1.1.0rc4-py2.7.egg/stetl/component.py", line 185, in process
    packet = self.next.process(packet)
  File "/usr/local/lib/python2.7/dist-packages/Stetl-1.1.0rc4-py2.7.egg/stetl/component.py", line 185, in process
    packet = self.next.process(packet)
  File "/usr/local/lib/python2.7/dist-packages/Stetl-1.1.0rc4-py2.7.egg/stetl/component.py", line 185, in process
    packet = self.next.process(packet)
  File "/usr/local/lib/python2.7/dist-packages/Stetl-1.1.0rc4-py2.7.egg/stetl/component.py", line 174, in process
    packet = self.invoke(packet)
  File "/root/git/etl/calibration.py", line 196, in invoke
    df = Calibrator.filter_data(df, filter_col, self.filter_alpha)
  File "/root/git/etl/calibration.py", line 272, in filter_data
    df[col] = Calibrator.running_mean(df[col], alpha)
  File "/root/git/etl/calibration.py", line 287, in running_mean
    val = elem * alpha + val * (1.0 - alpha)
TypeError: unsupported operand type(s) for -: 'float' and 'str'
```

pietermarsman, 2017-01-25:
Hey

Yes, kwam ik ook achter. Gebruikte per ongeluk een Config met type float.
Dat mag niet van stetl en dan maakt ie er een str van.

Is opgelost met de laatste commit.

Pieter.

closed 2017-01-26 by pietermarsman