smartemission/smartemission #48: Maak gebruik van temporele relatie tijdens calibratie

pietermarsman, 2017-02-05:
Ik heb de code voor low-pass filtering voor de calibratie (en voor toepassen van het calibratie model) in een comment gezet. Dit leverde veel bugs op omdat ik niet op lette. 

In de gas metingen (s_no2resistance, s_o3resistance, s_coresistance) zit ruis. Door een running/rolling mean te gebruiken kan dit eruit worden gehaald waardoor de calibratie beter wordt. Voor deze oplossing moet per device/geohash een rolling mean worden bijgehouden voor elk van de gassen.

Een andere oplossing is ook vorige inputs aan het calibratie model meegeven. Dan zou die de huidige RIVM meting kunnen voorspellen aan de hand van de afgelopen n (bijvoorbeeld 10) jose metingen. Hiervoor moet per device/geohash de laaste n metingen worden bijgehouden in een lijst. 

Nog een andere oplossing is een recurrent neural network. Zie ook #36. Deze heeft een staat en de volgorde waarin observaties worden aangeboden maakt dus uit. Dus moet er worden gegarandeerd dat observaties chronologische worden aangeleverd. En elke device/geohash moet zijn eigen kopie van het model hebben.

pietermarsman, 2017-02-05:
Door het verwijderen van de low-pass filter verslechterd de performance:
CO: 74% -> 57%
NO2: 78% -> 66%
O3: 96% -> 92%

pietermarsman, 2017-02-07:
Nog 1 probleem over: als de last en refiner services gewoon draaien hebben ze steeds maar een kleine hoeveelheid nieuwe datapunten per run. De RunningMean states gaan dan steeds verloren als het proces weer opnieuw wordt opgestart. Eigenlijk moet deze state worden opgeslagen. 

Ideeen om dit makkelijk te doen Just?

justb4, 2017-02-07:
Pickled objecten in als blob in Postgres DB  opslaan?

pietermarsman, 2017-02-07:
Dat is wel makkelijk :) 

Ik zal kijken of ik dit een beetje generiek kan implementeren. Ik denk dat ik in de toekomst wel met een andere representaties van de state van een device ga werken. Dus niet alleen running means, maar meer of anders. 

Ik ga dan denk een refiner-state (oid) pickled of jsonified opslaan in de db. De refiner-state bestaat dan weer uit een device-state en mogelijk meer.

pietermarsman, 2017-02-08:
Just, kun jij de laatste commit bekijken. Daarin is nogal wat veranderd en ik ben benieuwd of je problemen ziet die ik over het hoofd heb gezien. 

Om de statefull calibratie te gebruiken is een refresh van de calibrated db nodig: `db-init-calibrate.sh`

justb4, 2017-02-08:
Ja idd nogal veel :-), in de code zie ik niet direct problemen, maar dat moet blijken uit calibratie. Ik zal de calibrator en daarna refiner/last op test weer draaien.

commit 10d3c608 (2017-02-07): closes #48, using filtering during calibration
commit e7184041 (2017-02-07): issue closes #48, using filtering during calibration
closed 2017-03-30 by pietermarsman