smartemission/smartemission #49: InfluxDB crash - out of memory

justb4, 2017-02-06:
```
[2417457.246864] Out of memory: Kill process 18104 (influxd) score 757 or sacrifice child
[2417457.246875] Killed process 18104 (influxd) total-vm:9868536kB, anon-rss:7502348kB, file-rss:0kB
```

pietermarsman, 2017-02-06:
Dit heeft er denk ik mee te maken met de dashboards/queries waar ik op dat moment met grafana mee bezig was: http://test.smartemission.nl/grafana/dashboard/db/jose. Hoe laat gebeurde het ongeveer? Om 16:39 toverde grafana geen data meer tevoorschijn (maar ook geen error) en was mijn laatste bezoek.

In de bovenste grafiek wou ik voor alle device_id, voor alle componenten alle metingen zien, gegroepeerd per uur. Daar deed ie nogal lang over. Misschien kun je het daarmee testen.

justb4, 2017-02-07:
Aha dat is een goeie, ik dacht dat het aan de Refiner die bezig was, lag, maar was idd even later:

```
2017-02-06 16:49:10,936 RefineFilter INFO Result unique_id=23-20170130-1 gid_raw=197800 record_count=12 val_errs=1168
2017-02-06 16:49:11,772 dboutput INFO committed 12 records
2017-02-06 16:49:11,772 RefineFilter INFO processing unique_id=23-20170130-1 gid_raw=197800 ts_count=324
2017-02-06 16:49:12,202 RefineFilter INFO Result unique_id=23-20170130-1 gid_raw=197800 record_count=4 val_errs=577
2017-02-06 16:49:12,203 influxdboutput INFO Creating payload from 4 records
2017-02-06 16:49:12,203 influxdboutput INFO Created payload of 348 characters
2017-02-06 16:49:22,565 httpoutput INFO Req nr 27187 - response status: code=-1 msg=''
2017-02-06 16:49:22,567 httpoutput ERROR Headers: None
2017-02-06 16:49:22,584 dbinput INFO Exit: disconnect from DB
2017-02-06 16:49:23,090 dbinput INFO Exit: disconnect from DB
2017-02-06 16:49:23,407 dboutput INFO Exit: disconnect from DB
Traceback (most recent call last):
  File "/usr/local/bin/stetl", line 5, in <module>
    pkg_resources.run_script('Stetl==1.1.0rc4', 'stetl')
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 528, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 1394, in run_script
    execfile(script_filename, namespace, namespace)
  File "/usr/local/lib/python2.7/dist-packages/Stetl-1.1.0rc4-py2.7.egg/EGG-INFO/scripts/stetl", line 40, in <module>
    main()
  File "/usr/local/lib/python2.7/dist-packages/Stetl-1.1.0rc4-py2.7.egg/EGG-INFO/scripts/stetl", line 31, in main
    etl.run()
  File "/usr/local/lib/python2.7/dist-packages/Stetl-1.1.0rc4-py2.7.egg/stetl/etl.py", line 101, in run
    chain.run()
  File "/usr/local/lib/python2.7/dist-packages/Stetl-1.1.0rc4-py2.7.egg/stetl/chain.py", line 208, in run
    packet = self.first_comp.process(packet)
  File "/usr/local/lib/python2.7/dist-packages/Stetl-1.1.0rc4-py2.7.egg/stetl/component.py", line 185, in process
    packet = self.next.process(packet)
  File "/usr/local/lib/python2.7/dist-packages/Stetl-1.1.0rc4-py2.7.egg/stetl/component.py", line 185, in process
    packet = self.next.process(packet)
  File "/usr/local/lib/python2.7/dist-packages/Stetl-1.1.0rc4-py2.7.egg/stetl/splitter.py", line 57, in process
    comp.process(packet)
  File "/usr/local/lib/python2.7/dist-packages/Stetl-1.1.0rc4-py2.7.egg/stetl/component.py", line 185, in process
    packet = self.next.process(packet)
  File "/usr/local/lib/python2.7/dist-packages/Stetl-1.1.0rc4-py2.7.egg/stetl/component.py", line 174, in process
    packet = self.invoke(packet)
  File "/usr/local/lib/python2.7/dist-packages/Stetl-1.1.0rc4-py2.7.egg/stetl/output.py", line 22, in invoke
    packet = self.write(packet)
  File "/usr/local/lib/python2.7/dist-packages/Stetl-1.1.0rc4-py2.7.egg/stetl/outputs/httpoutput.py", line 177, in write
    self.post(packet, self.create_payload(packet))
  File "/usr/local/lib/python2.7/dist-packages/Stetl-1.1.0rc4-py2.7.egg/stetl/outputs/httpoutput.py", line 157, in post
    res = webservice.getfile().read()
AttributeError: 'NoneType' object has no attribute 'read'
```

justb4, 2017-10-03:
Voorlopig opgelost.

closed 2017-10-03 by justb4