smartemission/smartemission #50: Problemen met Whale server http opvragingen

justb4, 2017-03-29:
De laatste weken komen er errors uit de Whale server, m.n. bij opvragen timeseries, bijv
``http://<whale-url>//sensors/v1/devices/22/timeseries`` (`whale url` symbolisch):

```
{
Message: "An error has occurred."
}
```

Dit is niet permanent: een volgende aanroep kan weer sucessvol zijn:

```
{
id: "22",
days: [
"/sensors/v1/devices/22/timeseries/20160402",
"/sensors/v1/devices/22/timeseries/20160403",
"/sensors/v1/devices/22/timeseries/20160404",
"/sensors/v1/devices/22/timeseries/20160405",
"/sensors/v1/devices/22/timeseries/20160406",
"/sensors/v1/devices/22/timeseries/20160407",
"/sensors/v1/devices/22/timeseries/20160408",
"/sensors/v1/devices/22/timeseries/20160409",
"/sensors/v1/devices/22/timeseries/20160410",
"/sensors/v1/devices/22/timeseries/20160411",
.
.
```

Gebeurt ook bij andere aanroepen, niet alleen ``timeseries``. In ieder geval zorgt dit ervoor dat de Smart Emission Harvester en andere ETL processen, zoals "Last Values" niet "door kunnen lopen" bijv logfile:

```
2017-03-29 12:25:02,798 RawSensorAPI INFO processable hours for device 7 day 20160731: []
2017-03-29 12:25:02,798 RawSensorAPI INFO Init: fetching timeseries days list from URL: "http://<whale url>/sensors/v1/devices/8/timeseries" ...
2017-03-29 12:25:03,247 httpinput ERROR HTTPError fetching from URL http://<whale url>/sensors/v1/devices/8/timeseries: code=500 e=HTTP Error 500: Internal Server Error
2017-03-29 12:25:03,247 RawSensorAPI INFO Exit
2017-03-29 12:25:03,248 dboutput INFO Exit: disconnect from DB
Traceback (most recent call last):
  File "/usr/local/bin/stetl", line 5, in <module>
    pkg_resources.run_script('Stetl==1.1.0rc4', 'stetl')
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 528, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 1394, in run_script
    execfile(script_filename, namespace, namespace)
  File "/usr/local/lib/python2.7/dist-packages/Stetl-1.1.0rc4-py2.7.egg/EGG-INFO/scripts/stetl", line 40, in <module>
    main()
  File "/usr/local/lib/python2.7/dist-packages/Stetl-1.1.0rc4-py2.7.egg/EGG-INFO/scripts/stetl", line 31, in main
    etl.run()
  File "/usr/local/lib/python2.7/dist-packages/Stetl-1.1.0rc4-py2.7.egg/stetl/etl.py", line 101, in run
    chain.run()
  File "/usr/local/lib/python2.7/dist-packages/Stetl-1.1.0rc4-py2.7.egg/stetl/chain.py", line 208, in run
    packet = self.first_comp.process(packet)
  File "/usr/local/lib/python2.7/dist-packages/Stetl-1.1.0rc4-py2.7.egg/stetl/component.py", line 168, in process
    result = self.before_invoke(packet)
  File "/opt/geonovum/smartem/git/etl/rawsensorapi.py", line 568, in before_invoke
    self.next_hour()
  File "/opt/geonovum/smartem/git/etl/rawsensorapi.py", line 535, in next_hour
    self.next_day()
  File "/opt/geonovum/smartem/git/etl/rawsensorapi.py", line 521, in next_day
    self.fetch_ts_days()
  File "/opt/geonovum/smartem/git/etl/rawsensorapi.py", line 432, in fetch_ts_days
    json_str = self.read_from_url(ts_days_url)
  File "/usr/local/lib/python2.7/dist-packages/Stetl-1.1.0rc4-py2.7.egg/stetl/inputs/httpinput.py", line 93, in read_from_url
    raise e
urllib2.HTTPError: HTTP Error 500: Internal Server Error
```

RobertKieboom, 2017-03-29:
Er lijken issues met fragmentatie van de disks te zijn, waardoor timeouts kunnen optreden. Het is inmiddels enorm veel data die met redelijk hoge snelheid wordt geschreven. Wordt aan gewerkt, eerst maar eens een opschoning en een repair van de database...

RobertKieboom, 2017-03-29:
Daarnaast heeft iemand de directory compressed gemaakt, hetgeen niet goed is voor de performance van de database. Zeker niet gezien de relatief hoge snelheid van data throughput.

RobertKieboom, 2017-03-30:
Schijven zijn gecleand, proces loopt weer. Ziet er goed uit, er loopt op dit moment ook een backup (voor de zekerheid), dus de performance is wel iets minder dan normaal.

justb4, 2017-04-11:
Ik zie weer problemen met Whale server, connecties blijven hangen:

```
2017-04-11 20:44:11,213 httpinput ERROR URLError fetching from URL http://whale.citygis.nl/sensors/v1/devices/30/last: reason=[Errno 110] Connection timed out e=<urlopen error [Errno 110] Connection timed out>
```

justb4, 2017-04-12:
Pieter erbij om te laten weten dat dit a/d hand is op test en prod.

RobertKieboom, 2017-04-12:
We hebben updates geinstalleerd en disk opgeschoond. We bekijken het error log nog, eens kijken hoe het nu gaat.

justb4, 2017-04-12:
Ik ben iets anders op spoor: er lijken algemene HTTP connectie problemen te zijn in de test en productie VMs...Vanuit mijn lokale systemen zie ik geen probleem meer. Kan iets na een Ubuntu of Docker upgrade zijn geweest...

justb4, 2017-04-12:
Hmm, het wordt nog vreemder: het blijkt een connectie-probleem naar Whale server i.h.a. te zijn dat alleen vanaf de twee SE servers optreedt, zelfs naar de IIS pagina http://whale.citygis.nl . Nergens anders vandaan, ook niet van een andere VM zelfde provider. Hieronder een `stress.sh` script dat ik gebruikte op verschillende systemen om 1000x een URL op te halen:

- gaat op elk systeem goed met elke URL BEHALVE
- alle http://whale.citygis.nl/* URLs vanaf dus de 2 SE servers met IPs 89.31.100.234 (test) en 141.138.193.58 (prod) dan gaan eerste paar requests ok, maar blokkeert na 4 of 5e.

Er lijkt een soort selectieve blocking te zijn, maar waar gebeurt dat?

```
#!/bin/bash

i="0"

while [ $i -lt 1000 ]

do
echo -ne "${i}."
# url="http://172.17.0.3:8080/v1.0/Things"
# url="http://kademo.nl"
# url=http://whale.citygis.nl/sensors/v1/devices
url=http://whale.citygis.nl
# url=http://www.citygis.nl
wget -O bla.txt  $url > /dev/null 2>&1
echo -ne "${i}+ "

i=$[$i+1]
done
```

justb4, 2017-04-13:
Dus conclusie:

- het zit niet in de SE harvesting software
- gebeurt alleen vanaf 2 IP adressen (die harvesten) van VPS-provider, 1 niet

Dan vermoed ik toch dat er een soort IP-blocking aan de "Whale-kant" zit...

RobertKieboom, 2017-04-13:
Ik heb het geconroleerd, we maken geen onderscheid naar IP adres. Heel vreemd, voor ons is er geen verschil tussen de verschillende adressen. Bij ons is in ieder geval geen blocking ingericht..

justb4, 2017-10-03:
Is opgelost inmiddels.

closed 2017-10-03 by justb4