smartemission/smartemission #53: Platform draaien op OSX

pietermarsman, 2017-04-11:
Gaat dit veel problemen opleveren? Ik heb nu een mac...

justb4, 2017-04-11:
Direct op Mac beter niet. Ik heb ook Mac en doe dit als volgt: installeer sowieso HomeBrew (voor packages). Dan met HomeBrew VirtualBox en Vagrant, zie bijv http://sourabhbajaj.com/mac-setup/Vagrant/README.html. 

Gebruik de Vagrantfile https://github.com/Geonovum/smartemission/blob/master/Vagrantfile door in die dir te gaan staan en "vagrant up" te doen. Het meeste moet verder hier beschreven staan:
http://smartplatform.readthedocs.io/en/latest/installation.html#local-install

Ik werk al tijden zo: voordeel je systeem vervuilt niet, je kunt altijd schoon beginnen, systeem is identiek aan test/data servers, en je kan experimenteren. Via de Vagrantfile port-mappings kan ik met lokale browser bij alle services.
De ETL zelf draai/test ik wel lokaal vanuit IntelliJ ontwikkelomgeving maar zonder Docker. Heb ook geen Docker voor Mac maar altijd via VirtualBox. 
Succes!

justb4, 2017-04-11:
O ja, de hele installatie binnen de VirtualBox Ubuntu (ik gebruik 14.04, hoger nog niet getest) kun je doen met bootstrap script:
http://smartplatform.readthedocs.io/en/latest/installation.html#id2

pietermarsman, 2017-04-12:
Met Vagrant en VirtualBox werkt inderdaad goed! 

Maar tijdens de build scripts loop ik tegen een probleem aan: 

- bij het aanmaken van users en databases (`db-init-influxdb.sh`) zijn er niet de juiste permission.
- als ik in influx CLI in de influx docker kijk zijn er inderdaad geen users, en is het ook niet mogelijk om users aan te maken
- als ik de conf volume mapping voor `influxdb.conf` weghaal uit `run.sh` van influxdb dan worden de instellingen niet meegenomen. Als ik dan in de influx CLI kijk zijn er drie users: youruser, youruser_se en youruser_as.
- Deze users lijken verdacht veel op de users uit `example.args`. Alleen heb ik geen idee hoe die info in `run.sh` is gekomen. Ook als ik `example.args` verwijder of verplaats blijft influx de drie users kennen (na docker opnieuw opstarten zonder config). 

Super bizar! Hoe krijg ik het voor mekaar dat de `run.sh` van de service influxdb de admin goed kan aanmaken?

justb4, 2017-04-13:
Ik ben daar ook tijdje mee bezig geweest, maar lukte uiteindelijk wel, ook herinitialisatie. `influxdb.conf` is zeker nodig. Misschien moet daarin initieel ` auth-enabled = false` (en daarna weer `true`) maken en dan `db-init-influxdb.sh` draaien met dezelfde settings die ik je net per email heb gestuurd ? Ook de logfiles moeten nog gemapped zie ik.

pietermarsman, 2017-04-17:
Probleem gevonden: de oude users blijven bestaan in '${DATA_DIR}/meta'. Als de docker wordt gestopt en verwijderd blijven deze users bestaan in de vagrant vm. Het verwijderen van die users zorgt ervoor dat inlfux met 'auth-enabled = true' kan worden opgestart maar er toch een admin account kan worden aangemaakt. 

Is het een idee om ergens om een goed gemikte plek een 'sudo rm -r ${DATA_DIR}/meta' neer te zetten? Ik had dat al gedaan in de 'run.sh' van influxdb maar dat is volgens mij niet de bedoeling.

justb4, 2017-04-19:
Ik denk dat  in `db-init-influxdb.sh` eerst de admin user verwijderd moeten worden. Iets als:

```
query "DROP USER ${influx_admin_user}
```

pietermarsman, 2017-04-19:
In `${DATA_DIR}/meta` zitten denk ik nog meer settings die onthouden worden. Net als in de rest van `${DATA_DIR}` trouwens. Is het niet de bedoeling dat na `db-init-influxdb.sh` de hele influxdb leeg is en vanaf nul geinitialiseerd word? In dat geval zou het verwijderen van de `${DATA_DIR}` volstaan namelijk.

Ik denk dat het lastig is om alle settings die gewijzigd zijn (zoals het aanmaken van users) terug te draaien aan het begin van `db-init-influxdb.sh`.

pietermarsman, 2017-06-06:
Duurt even maar ik kan aan de slag :)

closed 2017-06-06 by pietermarsman

justb4, 2017-06-07:
Ok, goed, in toekomst moeten we minder "host-dependencies" incorporeren voor Docker deployment, bijv met Docker Volumes, `docker-compose` en [Jobber](https://github.com/blacklabelops/jobber-cron) ipv native `cron`. Doe ik in nu andere projecten ook bijv: https://github.com/geopython/GeoHealthCheck/tree/master/docker