smartemission/smartemission #55: Noise measurement

Nijmegen-Geluid, 2017-04-18:
Noise levels show a deviation of 10 dB less than the real actual measured noise level with calibrated equipment.
Probably a factor 10 is forgotten in the formula counting the decibel levels.

Peter

justb4, 2017-04-19:
Bedankt Peter, ik ga dat onderzoeken. Het algoritme staat [hier](https://github.com/Geonovum/smartemission/blob/master/etl/sensorconverters.py#L252). De definities (inputs) staan [hier](https://github.com/Geonovum/smartemission/blob/master/etl/sensordefs.py#L442). Dwz we gebruiken de Whale banden: `'v_audio0', 'v_audioplus1', 'v_audioplus2', 'v_audioplus3', 'v_audioplus4', 'v_audioplus5', 'v_audioplus6', 'v_audioplus7', 'v_audioplus8', 'v_audioplus9'`.
Ieder van deze inputs heeft weer 3 subbanden.

Gemiddelde bepaling:
*  eerst *Decibel(A) to Bel and then get "real" value (power 10)*: `math.pow(10, band_val / 10)`
* gemiddelde in Db: *average of "real" values and convert back to Bel via log10 and Decibel via *10* `math.log10(band_avg / float(band_cnt)) * 10.0`

Eerst wordt gemiddelde per band gedaan (vd 3 subbanden), dan gemiddelde over alle banden, dus gemiddelde van gemiddelden. Steeds met inachtneming bovenstaande conversies. Dus nooit worden logaritmische waarden gemiddeld (altijd eerst macht 10 van Db band of gemiddelde waarde gedeeld door 10).

Maar ik moet er ook weer induiken, is tijd geleden, de faktor 10 komt paar keer voor: Db naar Bel en waarde terug naar Db. Ik weet ook niet of Db(A) of Db waarden nog uitmaakt hierbij.

Plus: ik moet ervan uitgaan dat de Db(A) waarden uit de Whale server precies kloppen. Dat moeten we eerst vaststellen. Denk als volgt:

* geef jij een reeks sensor nummers en tijden met jouw metingen
* via de Whale API kan ik de ruwe waarden opvragen bijv:
[sensor 22 op 24 maart 2017 uur 11 in UTC dat is tussen 10:00 en 11:00 UTC](http://whale.citygis.nl/sensors/v1/devices/22/timeseries/20170324/11)

Nijmegen-Geluid, 2017-05-02:
Hoi Just,

Zie Forum/overlegtafels/calibratie/over tijd en waarnemingen.

De uitgevoerde meting is de volgende:

  *   De resultaten van de meting met geijkte apparatuur.
De gemelde tijd is de starttijd van de meting. Lokale tijd.
Gemeten equivalent gemiddeld geluidsnivo over één uur is LAeq is 58 dB(A).
———————-
Bruel & Kjaer
SLM Type 2236
SETTINGS:
————————
F 30-110 dB
RMS: A Peak: L
RECORD NO,:
————————
29 Nov 2016 11:59:47
Elapsed Time 0001:00:07
Pauses 0
Overload 0,0 %
MaxP 94,7 dB
MaxL 76,6 dB
MinL 39,5 dB
Leq 58,0 dB    <<<<<<<<<<<<<<<<<<<<<<  58 dB(A) over één uur
SEL 93,5 dB
LEPd (Te= 7h30) 57,7 dB
L1 67,0 dB
L10 62,5 dB
L95 44,0 dB

1 decibel is 0,1 bel. Maar de bel wordt niet gebruikt.
Optellen decibellen; 10xLOG(10^Y1/10 + 10^Y2/10 + …..)
We middelen dus niet.
Alle (29 stuks) 1/3 octaafbanden optellen met bovenstaande formule geeft het totale geluidsniveau.
Drie 1/3 octaafbanden optellen geeft één 1/1 octaafband.
De 9 octaafbanden optellen geeft weer het totale niveau.
Voor de middenfrequenties van de 1/3 octaafbanden zie bijlage.

Als je meer uitleg wilt kom ik wel even langs. Dan kijken we er even samen naar.

Gr. Peter

Met vriendelijke groet,

Peter van der Voorn
Specialist Geluid
Bureau Leefomgevingskwaliteit PK40
Postbus 9105, 6500 HG, Nijmegen
Korte Nieuwstraat 6, 6511 PP, Nijmegen
06 51 33 07 03
024 329 96 92
p.van.der.voorn@nijmegen.nl

Nijmegen-Geluid, 2018-01-25:
Zullen we een afspraak maken om samen het algoritme aan te passen?

justb4, 2018-02-16:
Closing this issue, as it is a duplicate of #88. Relevant info attached there.

closed 2018-02-16 by justb4