smartemission/smartemission #58: Make predictions per hour instead of per sample

pietermarsman, 2017-06-07:
Currently predicting a gas values for each sample. Making predictions based on hourly aggregates has several advantages: 

- aggregated query can be used to get the jose data into python, this reduces the memory need.
- the dataset becomes a lot smaller, which makes learning faster
- noise is removed from the observations by taking the average
- no need to keep track of a running mean
- easier to understand
- less complex

A couple of things need to be done to get this to work:

[ ] Use an aggregated query for the calibration proces
[ ] Remove the running mean in the calibration process
[ ] Check if `rivm` and `joseextract` use the same timezone
[ ] Do the visualizations still work?
[ ] Use average of the refiner proces to predict new values
[ ] Alter last process to keep track of the values in the last hour instead of a running mean

justb4, 2017-09-27:
Ok, by accident merged these changes on Test server: 

```
nusa:git just$ git pull
remote: Counting objects: 100, done.
remote: Total 100 (delta 30), reused 30 (delta 30), pack-reused 70
Receiving objects: 100% (100/100), 21.74 KiB | 0 bytes/s, done.
Resolving deltas: 100% (73/73), completed with 23 local objects.
From https://github.com/Geonovum/smartemission
   a845056..42dec87  master                     -> origin/master
 * [new branch]      feature_hourly_predictions -> origin/feature_hourly_predictions
Merge made by the 'recursive' strategy.
 etl/calibration_output.py      | 9 ++++++++-
 etl/db/db-schema-calibrate.sql | 2 ++
 etl/extractor.cfg              | 1 -
 etl/influxdboutput.py          | 7 ++++---
 services/gost/README.md        | 6 +++---
 5 files changed, 17 insertions(+), 8 deletions(-)
```

Maybe this issue was halfway? Started with upgrading the database table on Test with `calibration_parameters` with the 2 new fields (`search_id` and `setting_id`). Will run `calibrator.sh` and see further.

justb4, 2017-09-27:
`calibrator.sh` fails since there is no `geohash` column in `joseextract` only a `geohash_1` column:

```
SELECT * FROM joseextract WHERE geohash != NULL AND time > '1900-01-01T00:00:00Z' LIMIT 1000;
```
gives 0 results.

Also `rivm` has partly `geohash_1` columns filled. See image.

[image]

Strange, nowhere is the `geohash_1` column defined. Looks like something happening automatically in InfluxDB?

justb4, 2017-09-27:
Ok `geohash_1`, due to *The name and symbol columns have _1 because your points have both of these stored as a tag and a field.* as [here](https://github.com/influxdata/influxdb/issues/7454#issuecomment-253276412).

justb4, 2017-09-27:
So question is why `geohash` was changed from `tag` to `field` (data) via commit:
https://github.com/Geonovum/smartemission/commit/caf98e7d7aee12e15cd2fb92c62e57235eb3ffc2, changes as in image:

[image] 

(Right is latest version in `master`). Also on [L156](https://github.com/Geonovum/smartemission/blob/master/etl/influxdboutput.py#L156) I think the `geohash_template` should then not contain a `,` but as field should be defined as `self.geohash_template = ' geohash="%s"'`, as `fields` are space separated.

pietermarsman, 2017-09-27:
Hi Just,

I am on a vacation so I don't have the possibility to look at any code. I remember two things: this feature was almost done, but not completely. And, geohash was changed from tag to field for a reason, I can't however which one.

Possible reasons I can think of / remeber:

- string /regex search is only possible on fields
- grouping aggregates are only possible on fields

When using hourly averages for the computation Influxdb can be used to do the aggregation. But this  was trickier than I thought. Not sure if the 'geohash as field' change is still necessary though.

pietermarsman, 2017-09-27:
The description of #58 provides a little information of why I did it.

Since the geohashes can vary widely (when using a large precision like we do) using it as a tag makes less sense.

justb4, 2017-09-28:
Ok, let's leave this until you are back. As a thought: [Influx docs](https://docs.influxdata.com/influxdb/v1.0/query_language/data_exploration/#regular-expressions-in-queries) describe using regular expressions on `tags` as:

[image]

`Tags` are much more efficient in queries over `fields`. This could mean rough "bounding-box" queries e.g. store `geohash` with 12-char precision and query with 7-char (works in most parts of the world).

Other issue is that when switching from `tag` to `field` or reverse on an existing InfluxDB table, the existing name gets "mangled" resulting in `geohash_1` (see above) etc.

justb4, 2018-01-03:
Got time to spend on this issue. First getting the "geohash field or tag issue" fixed since the InfluxDB data is now "mixed" by name-mangling (see comment above). I prefer using as `tag` as to be able to spatially query data, even as regex.

I think this [email conversation ](https://groups.google.com/forum/#!searchin/influxdb/geohash%7Csort:relevance/influxdb/zq4n_VmyG2Q/rAOcPe6IAQAJ) (with the InfluxDB lead dev) and subsequent [open issue](https://github.com/influxdata/influxdb/issues/8219) summarizes the limitations of `geohash` as tag.

So IMHO:

* `geohash` as tag: can do `SELECT` on regex  of tag-value, e.g. within area
* what is lacking is the possibility for `GROUP BY` on `tag` value-substrings, hence I think @pietermarsman changed from tag to field (to be able to do aggregations on Jose minute-data)

I see several options:

* store high-precision geohash as `field` and lower precision (plm 20m) as `tag`
* let the `Extractor` do the aggregation on the raw data values explicitly, just like the `Refiner`, i.e. store aggregated/averaged raw hour-values in InfluxDB `joseraw` (just like RIVM). No need for aggregate/GROUP BY queries, can use geohashes just like now
* make the "merging" Jose-RIVM smarter: let the (aggregated) Jose-data be leading: querying the nearest RIVM station

Ok, has also be some time ago, need to get into all details again, hope @pietermarsman has some time to review and comment.

justb4, 2018-01-04:
Ok, for `geohash` the `InfluxDbOutput` class supports adding them as `fields` and/or `tags` via the config. The strategy for now is to add a 7-char precision `tag`, called `geohash_tag` and a 12-char precision `field` called `geohash`. Both the Extractor and HarvesterRIVM are using these now. The Calibrator and the `*.args` files have been adapted to reflect this in queries. The query by `geohash_tags` works. And even better, the `Calibrator` ETL-process is (almost) working again! Some error: 

```
2018-01-04 18:08:23,783 InfluxDbInput INFO Querying database: SELECT time, component, geohash, value FROM joseextract WHERE (geohash_tag='u1hnvkb' or geohash_tag='u1hnvkc' or geohash_tag='u1hnvm3' or geohash_tag='u178hw8' or geohash_tag='u178kt7') AND time > '2017-01-08T21:01:11Z' LIMIT 1000000
2018-01-04 18:08:24,007 InfluxDbInput INFO Received 10000 results
2018-01-04 18:08:24,011 InfluxDbInput INFO Last timestamp from influxdb is 2017-01-16T10:56:51Z
2018-01-04 18:08:24,041 InfluxDbInput INFO Querying database: SELECT time, component, geohash, value FROM joseextract WHERE (geohash_tag='u1hnvkb' or geohash_tag='u1hnvkc' or geohash_tag='u1hnvm3' or geohash_tag='u178hw8' or geohash_tag='u178kt7') AND time > '2017-01-16T10:56:51Z' LIMIT 1000000
2018-01-04 18:08:24,076 InfluxDbInput INFO Received 2096 results
2018-01-04 18:08:24,080 InfluxDbInput INFO Last timestamp from influxdb is 2017-01-16T15:59:52Z
2018-01-04 18:08:24,103 InfluxDbInput INFO Querying database: SELECT time, component, geohash, value FROM joseextract WHERE (geohash_tag='u1hnvkb' or geohash_tag='u1hnvkc' or geohash_tag='u1hnvm3' or geohash_tag='u178hw8' or geohash_tag='u178kt7') AND time > '2017-01-16T15:59:52Z' LIMIT 1000000
2018-01-04 18:08:24,107 InfluxDbInput INFO Received 0 results
2018-01-04 18:08:24,641 Calibration INFO Received rivm data with shape (78564, 5)
2018-01-04 18:08:24,641 Calibration INFO Received jose data with shape (736664, 10)
2018-01-04 18:08:24,641 Calibration INFO Pre-processing geohash and time
2018-01-04 18:08:34,725 Calibration INFO Interpolating RIVM values towards jose measurements
2018-01-04 18:08:44,418 Calibration INFO RIVM and Jose are merged, new shape (736738, 13)
2018-01-04 18:08:45,230 Calibration INFO After dropping, filtering and sampling a data frame with shape (4179, 11) is ready for calibration
2018-01-04 18:08:45,233 Calibration INFO Starting randomized cross validated search to find best parameters. Running 50 iterations with 10 cross validations of 1 cores. Finding relation from s_barometer, s_coresistance, s_humidity, s_lightsensortop, s_no2resistance, s_o3resistance, s_temperatureambient, s_temperatureunit to carbon_monoxide__air_.
2018-01-04 18:31:03,932 Calibration INFO Best result (0.54) obtained by using {'mlp__activation': 'tanh', 'mlp__alpha': 5.7206547320557027e-05, 'mlp__hidden_layer_sizes': 133, 'mlp__learning_rate_init': 1.7563437893099812e-06, 'mlp__solver': 'lbfgs', 'mlp__learning_rate': 'constant', 'mlp__early_stopping': True, 'mlp__max_iter': 200}
2018-01-04 18:32:47,349 Calibration INFO Returning packet of size 12
2018-01-04 18:32:47,355 dboutput INFO query is INSERT INTO calibration_models (parameters,n,score,predicts,input_order,model) VALUES (%s,%s,%s,%s,%s,%s)
2018-01-04 18:32:47,377 dboutput INFO query is INSERT INTO calibration_parameters (std_train_score,rank_test_score,mean_train_score,setting_id,parameter,std_test_score,mean_test_score,n,std_fit_time,predicts,value,mean_fit_time,search_id) VALUES (%s,%s,%s,%s,%s,%s,%s,%s,%s,%s,%s,%s,%s)

.
.
2018-01-04 18:32:47,735 dboutput INFO committed 400 records
2018-01-04 18:32:47,738 Visualization INFO Creating dir calibration/carbon_monoxide__air_
2018-01-04 18:32:47,740 Visualization INFO Visualizing error as scatterplot
2018-01-04 18:32:48,240 Visualization INFO Saved figure to calibration/carbon_monoxide__air_/error_scatter.png
2018-01-04 18:32:48,243 Visualization INFO Visualizing error as histogram
2018-01-04 18:32:48,541 Visualization INFO Saved figure to calibration/carbon_monoxide__air_/error_histogram.png
2018-01-04 18:32:48,544 Visualization INFO Visualizing time series 20170101 - 20170201
2018-01-04 18:32:48,734 Visualization INFO Saved figure to calibration/carbon_monoxide__air_/time_series_20170101_20170201.png
2018-01-04 18:32:48,737 Visualization INFO Visualizing time series 20170101 - 20170107
2018-01-04 18:32:48,911 Visualization INFO Saved figure to calibration/carbon_monoxide__air_/time_series_20170101_20170107.png
2018-01-04 18:32:48,915 Visualization INFO Visualizing time series 20170101 - 20170102
2018-01-04 18:32:49,083 Visualization INFO Saved figure to calibration/carbon_monoxide__air_/time_series_20170101_20170102.png

```

But the subject of this issue "predictions per hour" still needs to be done...

justb4, 2018-01-06:
Hmm, data is harvested again ok from RIVM (harvester_rivm) and SE raw data (extractor) but not all stations used for calibration provide (clean), see this issue #77.