smartemission/smartemission #62: Sensor Jose-060 working properly but data is not seen in any viewer

Nijmegen-Geluid, 2017-10-24:
Please check why data is not seen in the viewers. Sensor Jose060 is connected!
Wifi network is connected to Wifi station GN-Extern in the building of the local goverment of Nijmegen.

justb4, 2017-10-24:
http://whale.citygis.nl/sensors/v1/devices/60/last no data is received from Whale server for device 60,...

justb4, 2017-10-24:
See timeseries for station 60:  no data available in Whale server since 30 jan 2017:
http://whale.citygis.nl/sensors/v1/devices/60/timeseries
Maybe @RobertKieboom can check further.

Nijmegen-Geluid, 2017-10-31:
The same problem for the Sensor Jose063.
And is also connected by Wifi station GN-Extern.

justb4, 2017-10-31:
Same story: no data received by Whale server: http://whale.citygis.nl/sensors/v1/devices/63/last

Nijmegen-Geluid, 2017-11-06:
Sensor Jose-063 is working properly now.
But sensor Jose-060 is still not working. Hardware seems to work allright. But data is not shown in any viewer.

justb4, 2017-11-06:
Sorry, but this is somehting @RobertKieboom can only verify (see above):
no data available in Whale server since 30 jan 2017:
http://whale.citygis.nl/sensors/v1/devices/60/timeseries

justb4, 2018-02-08:
Sorry nothing we can do here to solve.

closed 2018-02-08 by justb4