smartemission/smartemission #78: Jose sensoren niet te zien op de SmartApp

Nijmegen-Geluid, 2018-01-25:
Jose sensoren 089 én 091 zijn niet te zien op de SmartApp.
Wel op de WhaleViewer.
Graag verhelpen.

justb4, 2018-01-25:
Best even zoeken: probleem is dat 89 en 91 geen GPS locatie afgeven, dan stopt de verdere verwerking binnen SE platform voor die sensor. Idd is er wel data uit Whale:

http://whale.citygis.nl/sensors/v1/devices/89/last
http://whale.citygis.nl/sensors/v1/devices/91/last

Maar je ziet al dat `s_latitude` en `s_longitude` de waarde `4294967295` hebben. Die is ongeldig. Misschien staat sensor binnen? In ieder geval is er nooit GPS afgegeven voor bijv 91 sinds deze opnieuw opgestart is op 23 jan 2018 (daarvoor 9 maanden geheel uit de lucht).
Dus probleem zal verholpen zijn als er weer GPS waarden afgegeven worden. 

Zelfde soort problemen middels issue #77. Voor de juiste verwerking moeten de benodigde indicatoren aanwezig zijn.

Nijmegen-Geluid, 2018-01-26:
Er lagen 5 Jose sensoren op mijn bureau. 3 waren wel te zien. Ze geven alle data goed weer. En kunnen elders worden ingezet.

Deze 2 hebben blijkbaar een minder goed of niet werkende GPS module. Zal ze eens bij het raam leggen.

Hoe lang duurt het ongeveer voordat een Jose sensor zichtbaar is in de SmartApp nadat die voor het eerst in het net hangt?

justb4, 2018-01-26:
@Nijmegen-Geluid Het SE platform leest elke 3 minuten de laatste waarden uit de Whale server, dus zou snel zichtbaar moeten zijn. Voor niet werkende GPS kun je beste Intemo contacteren.

Nijmegen-Geluid, 2018-01-26:
Jose sensor 091 doet het inderdaad netjes. Blijkbaar een iets zwakkere GPS module.
Jose sensor 089 niet. Gaat inderdaad terug naar Intemo. Met nog 4 andere sensoren die om andere redenen niet goed functioneren.
NB Het duurt veel langer dan 3 minuten voordat een Jose sensor op de SmartApp zichtbaar is. Eerder 3 uur!!

justb4, 2018-02-08:
Kunnen we deze issue sluiten?

closed 2018-02-08 by justb4