smartemission/smartemission #82: Ontbrekende CORS header in data.smartemission.nl services

haayman, 2018-02-07:
De diverse services op data.smartemission.nl zijn niet geschikt om vanuit ajax aan te roepen, omdat er geen CORS headers zijn gezet.

Verzoek om in de response-headers de volgende regel op te nemen:
Access-Control-Allow-Origin: *

justb4, 2018-02-07:
CORS headers are set for all web access on the http://data.smartemission.nl/ and http://test.smartemission.nl/ domains. Please close if ok after testing.

hylkevds, 2019-02-07:
https://data.smartemission.nl/gost/v1.0 does not return CORS headers (yet? any more?)

justb4, 2019-02-07:
@hylkevds good you found this issue. It was fixed/closed back then when running Apache as frontend. Now that we moved to PDOK/Kubernetes (sept 2018) we need to CORS headers again. @gerdos82 can you pick this up in PDOK?