smartemission/smartemission #88: dB(A)values in SmartApp and Heronviewer are not correct. L

Nijmegen-Geluid, 2018-02-14:
The values which are shown are 10 dB lower than they should be.

justb4, 2018-02-15:
Is lastig uit te vinden. @Nijmegen-Geluid Heb je vergelijkings materiaal: bijv dat een waarde uit de Whale en uit SmartApp? Er worden namelijk best veel bewerkingen uitgevoerd vanaf de ruwe data tot de waardes in de viewers. Er wordt veel gemiddeld, wel altijd met inachtneming dat dB's logaritmische eenheden zijn. Dus bij middeling wordt altijd macht10 gedaan en terug log10.

Dus ruwe waarde:
```
raw_val = math.pow(10, dba_val/ 10)
```
en terug

```
dba_val = log10(band_avg / float(raw_val)) * 10.0
```

Ik neem aan dat dit goed is. IK heb een uitleg over audio in de documentatie toegevoegd [zie deze link](http://smartplatform.readthedocs.io/en/latest/data.html#audio-data-josene).

Ik ben wel iets op het spoor: een dB(A) kan m.i. negatief zijn. Volgens de specs loopt de waarden uit de Jose Sound Pressure sensors van plm 25 tot plm 100 dB(A). Heel vaak is de dB(A) waarde van een band echter 0. Die wordt dan ten onrechte meegenomen in de middeling en kan er dus voor zorgen dat de gemiddelde waarde 'omlaag wordt getrokken'. Mogelijk betekent de waarde 0 "geen waarde". In ieder geval heb ik de min/max config aangepast. Er komen nu wel andere waardes uit, niet perse allemaal hoger. Welke waardes vergeleek je? Was het altijd precies 10dB verschil? 

Hieronder resultaat: links is "oude berekening" rechterkolom, nieuwe met uitsluiten van 0-waarden:

[image]

Draait op testserver (http://test.smartemission.nl rechts), dus te vergelijken met productie server (http://data.smartemission.nl, links).

justb4, 2018-02-16:
With the most recent updates the production server is now also using the new sound pressure calculations (skipping 0 values when averaging).

justb4, 2018-02-16:
Ok, ik zie nu dat er al een issue #55 hiervoor was. Daarin staat een belangrijke opmerking van @Nijmegen-Geluid 
```
Hoi Just,

Zie Forum/overlegtafels/calibratie/over tijd en waarnemingen.

De uitgevoerde meting is de volgende:

  *   De resultaten van de meting met geijkte apparatuur.
De gemelde tijd is de starttijd van de meting. Lokale tijd.
Gemeten equivalent gemiddeld geluidsnivo over één uur is LAeq is 58 dB(A).
———————-
Bruel & Kjaer
SLM Type 2236
SETTINGS:
————————
F 30-110 dB
RMS: A Peak: L
RECORD NO,:
————————
29 Nov 2016 11:59:47
Elapsed Time 0001:00:07
Pauses 0
Overload 0,0 %
MaxP 94,7 dB
MaxL 76,6 dB
MinL 39,5 dB
Leq 58,0 dB    <<<<<<<<<<<<<<<<<<<<<<  58 dB(A) over één uur
SEL 93,5 dB
LEPd (Te= 7h30) 57,7 dB
L1 67,0 dB
L10 62,5 dB
L95 44,0 dB

1 decibel is 0,1 bel. Maar de bel wordt niet gebruikt.
Optellen decibellen; 10xLOG(10^Y1/10 + 10^Y2/10 + …..)
We middelen dus niet.
Alle (29 stuks) 1/3 octaafbanden optellen met bovenstaande formule geeft het totale geluidsniveau.
Drie 1/3 octaafbanden optellen geeft één 1/1 octaafband.
De 9 octaafbanden optellen geeft weer het totale niveau.
Voor de middenfrequenties van de 1/3 octaafbanden zie bijlage.

Als je meer uitleg wilt kom ik wel even langs. Dan kijken we er even samen naar.
```

Komt er dus op neer dat middelen niet correct is maar banden moeten worden opgeteld! Dan moet idd gehele berekening herzien worden. Logisch natuurlijk dat de freq-banden moeten worden opgeteld, zijn a.h.w. ieder een bron en dragen bij tot totale geluidsnivo. Is er ook nog weging nodig of is dat al impliciet vanwege dB(A)?  Ik vond ook een voorbeeld die lijkt op jouw berekening:
http://www.sengpielaudio.com/calculator-spl.htm

justb4, 2018-02-16:
Op grond jouw opmerking en de links als http://www.sengpielaudio.com/calculator-spl.htm een nieuwe berekening ingezet. Draait nu op testserver: http://test.smartemission.nl.

Bestaat uit 3 stappen.

### 1. 1/3 octaaf subbanden optellen
Eerste stap is 1/3 octaaf sub-banden optellen. Mijn optelling is dan precies gelijk aan wat uit deze calculator komt: http://www.sengpielaudio.com/calculator-octave.htm
Bijv 33,33,32 opgeteld geeft 37.463.

### 2.  octaven optellen
Dan lijkt mij dat dan alle octaven opgeteld moeten worden? Heb ik uit bovenstaande totaal 10 octaaf-waarden:
[34.0, 42.0, 44.0, 41.0, 34.0, 36.0, 36.0, 36.0, 37.0, 37.0] dan is som en dus noiselevel 49 lijkt mij.
Dat komt ook uit calculator:

[image]

### 3. Uurmiddeling
Per uur zal er m.i. wel echt gemiddeld moeten worden. Dat gebeurde al middels een 'moving average functie' die de machten van 10 optelt en middelt en uiteindelijk weer de log10 maakt (met inachtneming factor 10 decibel-bel).

    # M = M + (x-M)/n
    # Here M is the (cumulative moving) average, x is the new value in the
    # sequence, n is the count of values. Using floats as not to loose precision.
    def moving_average(self, moving_avg, x, n, unit):
        if 'dB' in unit:
            # convert Decibel to Bel and then get "real" value (power 10)
            # print moving_avg, x, n
            x = math.pow(10, x / 10)
            moving_avg = math.pow(10, moving_avg / 10)
            moving_avg = self.moving_average(moving_avg, x, n, 'int')
            # Take average of "real" values and convert back to Bel via log10 and Decibel via *10
            return math.log10(moving_avg) * 10.0

        # Standard moving avg.
        return float(moving_avg) + (float(x) - float(moving_avg)) / float(n)

justb4, 2018-02-16:
@Nijmegen-Geluid Wat ik net nog zag: welke octaven mee te nemen: nu gaan 10 octaven mee,
maar zag ergens je opmerking dat je van 31.5Hz tot 8KHz gaat. Dan eigenlijk 9 octaven meenemen:

Nu:
```
 'v_audio0', 'v_audioplus1', 'v_audioplus2', 'v_audioplus3', 'v_audioplus4', 
'v_audioplus5','v_audioplus6', 'v_audioplus7', 'v_audioplus8', 'v_audioplus9'
```

Met `v_audio0: 0-40Hz` en `v_audioplus8: 5-10KHz`, `v_audioplus9: 10-20kHz`. 
Dus `v_audioplus9` zou weggelaten moeten worden.

BIj `v_audioplus9: 10-20kHz` weglaten iets lagere waarden  of moeten zelfs nog subbanden uit `v_audio0 (de 40Hz subband)` en `v_audioplus8 (de 10KHz subband)` worden weggelaten?

justb4, 2018-02-16:
Die laatste calculaties draaien nu op test, dus:

- `v_audioplus9` weglaten
- `v_audio0` (de 40Hz subband) en `v_audioplus8` (de 10KHz subband) weglaten

Dus dat alleen 31.5Hz tot 8KHz wordt meegenomen.

Nijmegen-Geluid, 2018-02-17:
Hoi Just,

Ik denk dat ik nu ongeveer dezelfde waarden ga meten met mijn gekalibreerde meter.
Ik hou je op de hoogte.

Bedankt!

Gr. Peter

Met vriendelijke groet,

Peter van der Voorn
Specialist Geluid
Bureau Leefomgevingskwaliteit PK40
Team Geluid & Lucht
Postbus 9105, 6500 HG, Nijmegen
Korte Nieuwstraat 6, 6511 PP, Nijmegen
M: 06 51 33 07 03
T: 024 329 96 92
E: p.van.der.voorn@nijmegen.nl<mailto:p.van.der.voorn@nijmegen.nl>
W: www.nijmegen.nl<http://www.nijmegen.nl>

justb4, 2018-02-20:
@Nijmegen-Geluid : is dit issue opgelost en kan ik het sluiten?

Nijmegen-Geluid, 2018-02-20:
Ik zie nog steeds te lage waarden in de SmartApp en Heronviewer.
Als ik inlog via http://smartemission.ruhosting.nl/

justb4, 2018-02-20:
Nee maar dat zijn waarden uit de productie server. Zie boven: ik had juist de testserver uitgerust met  de nieuwe berekeningen, dus nogmaals:

* http://data.smartemission.nl/smartapp (huidig)
* http://test.smartemission.nl/smartapp NIEUW

Je moet dus bij onderste kijken. De historie aanpassen wordt sowieso lastig. Maar vanaf afgelopen  vrijdag 1800 zou op testserver aangepast zijn.

Nijmegen-Geluid, 2018-02-20:
Gezien.
Ziet er realistisch uit.
Gaan dit nog wel controleren. Maar dit kan niet direct.

Je kunt nr 88 afsluiten. En uit de meting zal blijken of het nu wel of nog niet in orde is.
We houden je op de hoogte.

Bedankt voor het aanpassen!

Gr. Peter

justb4, 2018-02-20:
Ok, ook op productie uitgerold.

closed 2018-02-20 by justb4