smartemission/smartemission #90: Most efficient way to get all Observations of the last week?

haayman, 2018-02-16:
In my app I want to display the trends for all observations of say the last 2 weeks or 2 days.

My current setup is to ask all datastreams:
Things?$expand=Locations,Datastreams
and then per Datastream:
Datastream.Observations?$filter=phenomenonTime gt [date]

This results in a *lot* of request

Next I tried to get everything in one go:
`/Things?$expand=Locations,Datastreams/Observations&$filter=Datastreams/Observations/phenomenonTime gt ${params.date.toISOString()}`

Due to paging this also results in hundreds of requests with a lot of redundent data shared between the results that I have to merge into single Thing objects

In the SOS service I was able to do 1 'stations?expanded=true' request and then per timeserie (==Datastream) a request like 'timeseries/{id}/getData?timespan={timespan}

Somehow this was way more efficiënt and fast. 

Do you have any tips to obtain this same efficiency?

justb4, 2018-02-16:
@bertt @tebben do you have suggestions? Would it help to increase the page size (`maxEntityResponse` now 100) to maybe 1000 or 10000?

hylkevds, 2019-02-27:
The query you wrote:
`/Things?$expand=Locations,Datastreams/Observations&$filter=Datastreams/Observations/phenomenonTime gt ${params.date.toISOString()}`
does not actually do what you think it does.

This gives you the Things that have an observation in the given time range. What you actually want is:
`/Things?$expand=Locations,Datastreams($expand=Observations($filter=phenomenonTime gt ${params.date.toISOString()}))`

The expand itself can take all query options, including expand. So here you expand the Datastreams, then for each Datastream, expand the Observations, but filtered on phenomenonTime.

Note, to reduce the response size, you can also add $selects:
`https://beaware.server.de/SensorThingsService/v1.0/Things?$select=name,id&$expand=Locations($select=name),Datastreams($select=name,id,properties;$expand=Observations($select=result,phenomenonTime;$filter=phenomenonTime%20gt%20now()%20sub%20duration%27P1D%27))&$skip=10`
I've skipped the first few Things, those have no observations