smartemission/smartemission #91: SmartApp Enhancements

geurp0, 2018-02-19:
The values of the Air-quality sensors don't make sense for the average user. Is it possible to add a colorscheme to the values indicating the quality of the value? Color scheme info to get at Henk Nijhuis

justb4, 2018-02-19:
Ok, see also [issue 64 comment](https://github.com/Geonovum/smartemission/issues/64#issuecomment-341756593): these color scheme's are already implemented in Heron (via WMS SLDs) with SLDs from RIVM. 

I propose to use the same color schemes here, otherwise things may get confusing. 
But ok, can you attach the Nijmegen color scheme here?

This is the proposed simple 5-color scheme (from RIVM):

[image]

corresponding to:

[image]

The SLDs use a more elaborate color scheme (from RIVM), see also https://www.luchtmeetnet.nl/uitleg#:

[image]

geurp0, 2018-02-19:
This colorscheme is ok. There is no Nijmegen schema :-)

justb4, 2018-02-20:
Welk kleurenschema te gebruiken: met 5 kleuren, zoals:
[image]
of met 11 zoals boven?

Voor eenvoud en begrijpelijkheid stel ik 5 kleuren voor.

justb4, 2018-02-20:
@geurp0 begonnen, in eerste aanzet kan zo bijv weergegeven:

[image]

of zo

[image]

Er zijn alleen kleurenschema's (van RIVM) voor NO2, O3, PM10 en PM2.5, niet voor CO en CO2, ook niet ergens kunnen vinden.

Er zal nog wel een legenda bij moeten, iets bedenken waar. Er gaat sowieso veel ruimte verloren aan logo's en plaatjes bovenin. Dat zou 1/2 tot 2/3 compacter kunnen.

geurp0, 2018-02-20:
5 kleuren graag. Kijk maar wat je aan overhead kan verminderen.

Paul Geurts
Strategisch Adviseur Informatie Management
Gemeente Nijmegen
06 – 53678077
p.geurts@nijmegen.nl
bereikbaar op maandag, dinsdag, woensdag, vrijdag

[image]<https://user-images.githubusercontent.com/582630/36439132-384fdd10-166c-11e8-913b-8974203f4eec.png>

[image]<https://user-images.githubusercontent.com/582630/36439172-55eee1ae-166c-11e8-8635-2345ef50def7.png>

geurp0, 2018-02-20:
Je zou locatie en laatste meting nog iets kleiner kunnen maken en dan 1 horizontale legenda maken die van slecht naar goed loopt. Dat kost ook niet zoveel ruimte. Idee?

Paul Geurts
Strategisch Adviseur Informatie Management
Gemeente Nijmegen
06 – 53678077
p.geurts@nijmegen.nl
bereikbaar op maandag, dinsdag, woensdag, vrijdag
________________________________
From: Paul Geurts
Sent: Tuesday, February 20, 2018 9:10:50 PM
To: Geonovum/smartemission; Geonovum/smartemission
Cc: Mention
Subject: Re: [Geonovum/smartemission] SmartApp: Add color scheme to Air Quality values (#91)

5 kleuren graag. Kijk maar wat je aan overhead kan verminderen.

Paul Geurts
Strategisch Adviseur Informatie Management
Gemeente Nijmegen
06 – 53678077
p.geurts@nijmegen.nl
bereikbaar op maandag, dinsdag, woensdag, vrijdag

[image]<https://user-images.githubusercontent.com/582630/36439132-384fdd10-166c-11e8-913b-8974203f4eec.png>

[image]<https://user-images.githubusercontent.com/582630/36439172-55eee1ae-166c-11e8-8635-2345ef50def7.png>

justb4, 2018-02-21:
Ja goede suggestie. Het SE logo is ook wel erg breed. Zou mooi zijn om iets kleins vierkant te hebben zoals ik in  favicon (SE webapps) gebruik, bijv:

[image]

Kan de ontwerper dat misschien doen? Het logo is nogal verwarrend ook: lijken net knoppen. Dit lijkt ook wat een smiley :-).
 
Wilde nog iets met een image rollover (hover) doen, maar wordt lastig met touch/mobiel.

Sowieso loop ik wat tegen het feit aan dat de SmartApp destijds ietwat quick & dirty is gebouwd...maar goed, kleuren en legenda moet gaan lukken.

justb4, 2018-02-21:
@geurp0 heb de hele popup compacter gemaakt, en AQ kleuren+legenda toegevoegd:

[image]

of direct via http://data.smartemission.nl/smartapp (evt expliciet herladen ivm browser caching).

geurp0, 2018-02-22:
Hoi just
Super gedaan. Heel intuïtief en geeft nu zoveel meer info.

Paul Geurts
Strategisch Adviseur Informatie Management
Gemeente Nijmegen
06 – 53678077
p.geurts@nijmegen.nl
bereikbaar op maandag, dinsdag, woensdag, vrijdag

[image]<https://user-images.githubusercontent.com/582630/36506726-c022a812-1757-11e8-8bb2-6ba6749cf6cd.png>

justb4, 2018-02-22:
Kunnen we deze issue dan sluiten? Via ander issue zou ik graag een nieuw SE logo zien, zie issue #95.

justb4, 2018-02-22:
Laatste wijziging: station nummer in popup heading en logo/SE naam vast rechtsboven in kaart, nog iets ruimte teruggewonnen..

[image]

geurp0, 2018-02-23:
Deze kan gesloten worden.

closed 2018-02-23 by geurp0