smartemission/smartemission #93: Smart App enhancement (3)

geurp0, 2018-02-19:
Is it possible to add new functionality to the SmartApp for Air-Quality  to show a diagram of the last 6 hours with the houraverages of the sensor. This gives the user relatively insight in the data. Is it getting better or worse?

justb4, 2018-02-19:
Ok, everything is possible :-). I would only remind that issue #60 was opened where specifically the Heron Viewer was chosen (by Nijmegen citizens and project members, though I was not there) for the main/all further viewer development. Issue #60, more of a "meta-issue", as it contains lots  of issues (will open one GH issue for each) in fact has been chosen as the basis for the enhancements internally known as "Annex I".

So a bit surprised, but on the other hand the SmartApp is more modern/mobile. It is more a matter where to put priorities (Heron or SmartApp Viewer enhancements/fixes). @geurp0 can you let know?

geurp0, 2018-02-19:
Just, we are now working on a specific scenario on the Waalkade. We will place a number of sensors along the river Waal. On the lamppoles we will place a sticker with a QR code, directly to the sensor in the SmartApp. The Smart App has an other target group than the citizen scientist. With the SmartApp, and the use on the street we hope that we can achieve that people get more interested on Air Quality. So there is a role for both platforms. There is a congres at the April 12. At that time the functionality should be active in the Smart App.

justb4, 2018-02-26:
@geurp0 Ok, thanks for clarification. There is a similar feature in the TNO App (Nijmegen Dashboard), image with graph showing history. Is there a way that viewer code can be shared to avoid duplicate efforts? I've created https://github.com/smartemission as a GH org to share multiple SE reps (we will split platform at PDOK as well) and e.g. viewers.

One other point: up to now the SmartApp has been interfacing only the "Last Values" SOSEmu API, a custom development at the start of the SE project. Currently we put effort on SensorThings API (STA), so we need to migrate the SmartApp to STA as well.