smartemission/smartemission #96: Migrate Josene devices from CityGIS Whale server to Intemo Collector

justb4, 2018-02-27:
CityGIS Whale server will be phased out and (Josene) devices will eventually send their data to the  Intemo Collector server. The migration will be one by one. As a device may be present on 2 servers at the same time, we need to skip those on the CityGIS Whale server.

Initial batch of device id's is 22,89,92,102,126,133. On feb 27 2018 14:30 UTC.

More device id's will be attached here.

avdcruyssen, 2018-04-03:
Sensors 36 and 53 are migrated to sensor1.intemo.com

justb4, 2018-04-04:
Is migration now complete and can we close this issue?

justb4, 2018-06-06:
Ok, will add 67 to be skipped from Whale server. 

@avdcruyssen are there more migrations planned? Or can we close this issue?

avdcruyssen, 2018-06-06:
More will follow, it's unclear which ones. All sensors I get for repair or service will be migrated.

avdcruyssen, 2018-06-29:
Sensor 9 (early prototype) just moved from Whale

justb4, 2018-08-03:
@avdcruyssen are we complete now and can this issue be closed?

avdcruyssen, 2018-08-10:
Sensors 20, 60 and 63 just moved to Intemo server

avdcruyssen, 2018-08-20:
Sensor 47, just moved to Intemo server

avdcruyssen, 2018-08-22:
Sensor 45, just migrated

avdcruyssen, 2018-08-27:
Sensors 13, 15 and 26 just migrated

avdcruyssen, 2018-08-28:
Sensor 17, just migrated

avdcruyssen, 2018-09-21:
Sensor 84, just migrated

avdcruyssen, 2018-11-05:
Jose 066, just migrated. Peter van der Voorn is going to migrate more sensors today, perhaps you should wait till tomorrow and migrated all at once to save work.

avdcruyssen, 2018-11-05:
Jose 044, just migrated

justb4, 2018-11-07:
@avdcruyssen ok thanks, @Nijmegen-Geluid can you let us know which sensors have migrated? Or is this the list: `66, 44, 23, 27, 32, 46, 16` (as above)? Also: can we phase out (harvesting from ) the CityGIS Whale server then? In the latest SmartApp, you can see using the Layers dropdown widget upper right, 
[image]
to show active sensors per project. Project is based on first 4 digits sensor id:

* no or 0000--> SE Nijmegen
* 2008 --> Green Capital Nijmegen
* 1182 --> AirSensEUR
* rest 2* --> Smart City Living Lab

See https://data.smartemission.nl/smartapp/.

avdcruyssen, 2018-11-07:
Numbers 33, 55, 30, 25, 62 migrated.

avdcruyssen, 2018-11-13:
Numbers 52, 54, 57 migrated

avdcruyssen, 2018-11-19:
Numbers 31, 87, 42, 61, 59, 50, 35 migrated

avdcruyssen, 2018-11-19:
Unsure, Whale server is not responding. Peter is still busy migrating, but most of those sensors are not in Nijmegen. I think maybe it is better to simply disable whale server in your services once Peter is finished. Anyway, they won't keep whale online, and it's not working at the moment at all so...

avdcruyssen, 2018-11-19:
Number 111 migrated

avdcruyssen, 2018-11-26:
Number 91 migrated

avdcruyssen, 2018-11-30:
Number 88 migrated

justb4, 2018-12-03:
Only station 48 now gives data from Whale Server: http://whale.citygis.nl/sensors/v1/devices/48/last  op coordinaten 48.200723, 16.36044. Is ergens in Wenen.

justb4, 2019-02-06:
@avdcruyssen  Are we all done? All sensors migrated? Then I can switch off all CityGIS harvesting.

avdcruyssen, 2019-02-07:
I think we are, haven't heard from Peter for a while. I'll ask him.

avdcruyssen, 2019-02-19:
Sensor 86, just migrated

avdcruyssen, 2019-05-03:
Sensor 18, just migrated