smartemission/smartemission #100: SE Nijmegen: pagina voor slimme Waalkade

justb4, 2018-03-26:
Uit mails met gem Nijmegen:

"... 8 tal sensoren op de Waalkade om de uitstoot van scheepvaart en het effect van walstroom te meten. Met de huidige viewer zien we dit niet zo snel. Voorbeeld hoe graag te zien:

[image]

Deze pagina willen we dan graag in een i-frame in onze website pagina voor dit project willen includen.

Het gaat om de sensoren met de range sensoren 20080001 t/m 20080008 

Het kaartje vervang ik nog met een andere als we definitief weten waar welke sensor komt  te staan. Interactie tussen kaart en sensoren is niet nodig. Het gaat om een oogopslag waar je snel alle last-values van de sensoren kan zien verdeeld over de waalkade.

Wat nog mooier zou zijn, maar dan wordt het denk ik toch wel complexer en duurder. Er is een web-service met al het scheepvaartverkeer. Je zou de varende schepen en de  stilliggende schepen op zich nog kunnen afbeelden in het kaartje, maar goed dan wordt het waarschijnlijk duur."

justb4, 2018-03-26:
eerste testversie met live data: http://test.smartemission.nl/waalkade 👍 
[image]

geurp0, 2018-03-27:
Hoi Just, 
Mooie eerste resultaat. Wat opmerkingen.
De pagina komt te draaien binnen greencapital2018.nl. Deze site is responsive en deze pagina (nog) niet. We moeten even kijken of dit mogelijk is. 
De sensoren op de Waalkade meten:
-GPS locatie 
-Temperatuur 
-Relatieve vochtigheid 
-Acceleratie 
-Licht intensiteit 
-CO2 -Geluid (31 banden) -Fijnstof sensor voor het indicatief meten van PM1, 2.5 en 10.

geurp0, 2018-03-27:
Nog een opmerking. We plaatsen één sensor waar we windrichting en windsnelheid meten. Zou je deze boven de tabel kunnen opnemen? Ik denk aan een windvaantje met hierbij de windrichting ( bv NW) en de snelheid. Stem je even met Janus af waar je die info vandaan krijgt?

geurp0, 2018-03-27:
De OV-masten op de waalkade worden voorzien van een sticker. Hierop staat het stations nummer een QR code  die doorlinkt naar de Smart Map viewer. Zie bijgevoegde voorbeeld sticker.
[QR1.pdf](https://github.com/Geonovum/smartemission/files/1851225/QR1.pdf)

justb4, 2018-03-27:
Ok, bedankt @geurp0. Antwoorden:

- het design maakt gebruik van Bootstrap responsive tables, maar er is een limiet wanneer de tabellen te breed zijn.
- indicatoren: PM10, PM2.5, Geluid, Temp en HUM: ok, PM1 wordt nog niet door systeem heen doorgegeven, kost extra tijd: ETL, kleurcoderingen (RIVM heeft ook geen PM1 m.i.)
- sensor nrs 1 t/m 8 ok (heb nu willekeurige genomen)
- link legenda ok
- meteo: wind(-richting) wordt nog niet doorgegeven, beter via andere API bijv KNMI of Wunderground, of derden die KNMI Live data aanbieden bijv: http://weerlive.nl/api/json-10min.php?locatie=51.85,5.86

geurp0, 2018-03-27:
Prima, 
Dan laten we PM1 even zitten, scheelt ook meteen weer in de breedte van de tabel, misschien gaat het dan passen. 
Voorlopig wind via een andere bron is ook goed. Het is allemaal toch redelijk indicatief.

justb4, 2018-03-27:
[image]

@geurp0 Ok, behalve wind meeste zaken toegevoegd, past nu ook beter mobiel, mogelijk nog (datum) tijd ook? Sensoren zijn: 20060001, 20060002, 20060003, 20060004, 20060005, 20060006, 20060007, 20060008 (Dordrecht) voorlopig.

geurp0, 2018-03-27:
Super! Wat zijn de voorkeur dimensies voor het kaartje, zodat hij optimaal mobiel te gebruiken is? Dan laat ik er eentje maken met die dimensies.
op mijn mobiel wordt de legenda onderaan opeens heel erg dik. Gaat hier iets fout?

geurp0, 2018-03-27:
Oh ja, datum en tijd toevoegen is een goed plan. Als mensen een schermafdrukje maken is dit erg handig. Wat mij betreft ook boven de tabel.

justb4, 2018-03-27:
Nieuwe versie:

* datum/tijd toegevoegd (nu bovenaan, kan ook elders, layout is met JS-templating)
* auto-refresh (alleen interne content) iedere 90 sec
* legenda tabel:  layout updates

Dimensie kaart: lastig, "mobiel" is erg breed begrip. Ik kan elk kaartje altijd verkleinen.
Ben ook niet van oorsprong designer ;-). De [App is gebaseerd op HTML5, Bootstrap4 met JQuery en Handlebars JS client-side templating](https://github.com/Geonovum/smartemission/tree/master/services/web/site/waalkade), dus layout refinements zou met hulp designer snel kunnen.

Wat wel nu punt is: net als SmartApp op "sosemu" API gebaseerd. Die zijn we aan het uitfaseren (bij PDOK), net als SOS, ten gunste van SensorThings API.

justb4, 2018-03-27:
* weer (wind) data van KNMI API toegevoegd, tabel compacter, data vermelding, animatie op verversen (iedere 1.5 min).

[image]

geurp0, 2018-03-28:
Hoi Just, bedankt voor de snelle samenwerking. Perfect zo. Migratie naar Sensorthings API zullen we komende tijd apart bespreken. Misschien in overleg met Kadaster.

geurp0, 2018-03-28:
De site staat nu op test.sm... etc. Wat is de definitieve URL voor de prod site?

geurp0, 2018-03-28:
Zojuist bericht van Janus gehad dat sensoren actief zijn:

Waalkade sensoren zijn publiek te zien op Smart Emission platform
http://smartemission.ruhosting.nl/viewers/
en op onderstaande viewer
https://josene.intemo.com/
 
ID nummers 2008000x

justb4, 2018-03-28:
Ja handig dat GitHub ;-). Nog niet alle stations zichtbaar. Zal definitief gaan draaien op http://data.smartemission.nl/waalkade . kan ergens deze week.

geurp0, 2018-03-28:
Ze zijn ze nu aan het ophangen, rond 13 uur hebben ze allemaal data.
We hebben een webcam stream die over de waalkade kijkt. 
http://app4.nijmegen.nl/gns/no_index/webcams/waalkade.html 
Ik stuur je zo de code en een static html pagina. Kan je die nog onder de tabel zetten?  Dan kunnen mensen zien hoe het meet de scheepvaart is gesteld is t.o.v. de metingen. Ik ben nu bezig om de webcam naar het westen te laten kijken.

geurp0, 2018-03-28:
https://www.nijmegen.nl/gns/no_index/webcams/waalkade2018.html
is een kale html-pagina

In de bron daarvan zie je het adres van de stream

https://5a5d6a3ee3a13.streamlock.net/live/waalkade.stream/playlist.m3u8

justb4, 2018-03-28:
Je bedoelt echt de video stream in de pagina? Gedaan, er zaten wel wat fouten in 
https://www.nijmegen.nl/gns/no_index/webcams/waalkade2018.html (deadlinks aantal CSS en JS imports), maar die rechtgestreken in http://test.smartemission.nl/waalkade/. Qua layout oogt de pagina wel wat rommelig nu. Misschien webcam dimensie kleiner. Ook de refresh moet niet mee.

geurp0, 2018-03-28:
Op de mobiel ziet het er wel aardig uit. Laat het maar even zo staan. Vanmiddag hebben we projectgroep dan gooi ik het even in de groep.

justb4, 2018-03-28:
We zijn druk op de testserver bezig ivm PDOK migratie, deze kan af en toe wegvallen, daarom de site vast op productie server gezet: http://data.smartemission.nl/waalkade .

justb4, 2018-03-28:
Ook nog: alleen dynamisch data-deel pagina wordt nu elke 90 sec ververst, niet bijv kaart en video.

geurp0, 2018-03-28:
[image]
[image]
Hierbij de nieuwe kaartjes. Eentje in de breedte van het webcam frame (640). De andere wat groter. Ben benieuwd hoe het eruit ziet.

justb4, 2018-03-28:
Ok, heb de grote kaart genomen, image responsive gemaakt, dus kaartje moet meeschalen.

geurp0, 2018-03-30:
Just, de temperatuur liep gisteren behoorlijk op. Het leek zomer in Nijmegen :-) Janus melde dat je de external temperatuur ipv ambient temp moet opnemen in de tabellen. Kan je deze aanpassen?

justb4, 2018-04-02:
Hm, dit is de eerste keer dat ik over deze indicator hoor, zie idd `s_externaltemp` uit de nieuwste devices via de Intemo API komen, maar bijv niet uit de "oude" (Whale/CityGIS) devices (is `s_temperatureambient`. Een wat indirecte manier om fundamentele wijzigingen in specs te communiceren..., maar we gaan kijken...

justb4, 2018-04-03:
ok afgerond, wordt apart issue voor `s_externaltemp`.

geurp0, 2018-04-09:
Just, 

We kunnen de pagina niet embedden omdat het geen http**s** is. Dat geldt trouwens ook voor de smartApp. DIt is eigenlijk wel de standaard tegenwoordig. Kan je deze ( en ik denk heel data.smartemission.nl) op http**s** zetten?

justb4, 2018-04-09:
Dat is best nog een flinke operatie. Voorheen hadden we daar een wildcard certificate voor nodig gehad. Binnen de SE PDOK migratie hebben we al wel https maar dat is op Kubernetes (dus niet te "backporten") en voorlopig nog niet vrijgegeven. FF overleggen hoe verder.

justb4, 2018-04-10:
De huidige omgeving wordt momenteel component voor component overgezet naar PDOK (Kubernetes K8s Cloud) in een "microservices architecture". Daar is automatisch nu al HTTPS (alles loopt via `nginx` K8s Ingress proxy). Om huidige en nieuwe omgeving dezelfde code te laten blijven gebruiken (anders 2 forks) gebruiken we [Docker Hub om alle componenten als Docker Images](https://hub.docker.com/u/smartemission/) te hosten. Met name de nu monolitische Apache App component wordt opgesplitst (smartapp, waalkade, heron etc). Dit alleen al kan heel soms verstoring geven in huidige omgeving. Maar voordeel is dat beide omgevingen voortdurend van dezelfde uitbreidingen/bugfixes profiteren.

Lang verhaal, maar punt is: HTTPS in huidige omgeving toevoegen heeft 2 nadelen:

* kans op verstoringen
* zou toegevoegd moeten worden aan huidige Apache die nu uitgefaseerd wordt

Voorstel is om via PDOK te hosten. Ook daar risico op verstoring omdat de omgeving nog niet officieel productie is. Evt problemen die we oplossen komen dan direct ten goede aan project. Dus  afgewogen beter dan huidige omgeving "patchen".

justb4, 2018-04-10:
Ha, we hebben de App naar PDOK gemigreerd en is nu achter https:
https://pdok.smartemission.nl/waalkade

geurp0, 2018-04-11:
Just, bij sensor 5 hangt de Pamares. Dit ding vangt fijnstof af. Hij werkt best goed want in de viewer zie je bij de PM waarden een streepje. Echter het streepje geeft verwarring. Kan je deze een waarde geven. Bv 0 ipv een streepje als de waarde dicht bij 0 ligt. Het lijkt nu alsof er niet gemeten wordt. Alvast bedankt.

justb4, 2018-04-11:
Ok. Andere vraag: doet de embedding met HTTPS het nu?

justb4, 2018-04-11:
Ik lees je vraag *"Hij werkt best goed want in de viewer zie je bij de PM waarden een streepje. "* en snap niet goed wat de bedoeling is. Een streepje betekent dat de app geen waarde van de server ontvangt. "de Pamares" heeft toch niets met de Josene te maken? Maar goed, ik krijg ook niet alle info, kan "0" geven als er niets ontvangen wordt.

geurp0, 2018-04-11:
Ik bedoel ermee dat de Pamares het goed doet. Hij vangt zoveel fijnstof af dat er geen fijnstof meer in de sensor terecht komt. De Josene viewer toont wel data: https://josene.intemo.com/sensors/last/118/ 
Allen de Smart App en en waalkade pagina geven geen resultaat. daar zit dus ergens iets in de overdracht en verwerking van de gegevens.

justb4, 2018-04-11:
Snap het nog steeds niet: *Hij vangt zoveel fijnstof af dat er geen fijnstof meer in de sensor terecht komt.* Dan doet ie het toch niet goed (zie bijv 0.02 ug/m3, dat is nagenoeg 0. In ieder geval stond de limiet op 1, waardoor er geen data naar de DB werd geschreven. Heb als test op onderlimiet op 0 (ug/m3) gezet. Zie http://data.smartemission.nl/waalkade/ Nu staat ie idd op 0, maar dan lijkt het toch alsof er geen fijnstof is? Terwijl eigenlijk de sensor niet werkt op dat moment, vandaar geen meting en dus '-'.

geurp0, 2018-04-11:
Volgens mij heb je het goed begrepen. Doordat je de limiet naar 0 hebt geschreven komen deze metingen ook binnen en ziet de gebruiker een 0. Dit was precies de bedoeling. De sensor werkt wel degelijk alleen de waarde is erg laag.

justb4, 2018-04-11:
Ok, temperatuur van KNMI voorlopig: eerst test: http://test.smartemission.nl/waalkade/

geurp0, 2018-04-11:
Prima zo, snel door naar productie. Embedden is inmiddels ook gelukt.

justb4, 2018-04-11:
Is ook gedaan. Echter, moet wel expliciet browser herladen. Dus nieuwe versie op https://pdok.smartemission.nl/waalkade/ heeft:

* temperatuur van KNMI op locatie
* 0 als minimum fijnstof

geurp0, 2018-04-11:
top!  bedankt

justb4, 2018-04-11:
voorlopig sluiten

closed 2018-04-11 by justb4