smartemission/smartemission #101: Time in smart app incorrect

Hans-Nouwens, 2018-03-29:
Time shown in smart app is not right.
'+2 hours' is not necessary 
[image]

justb4, 2018-03-29:
Well, the time, here 12:56:58 should be correct. The +02:00 denotes the offset from GMT/UTC including daylight savings. 12:56:58+02:00  is the standard [ISO 8601 ](https://en.wikipedia.org/wiki/ISO_8601) notation.

Hans-Nouwens, 2018-03-29:
For inhabitants the notation is not clear:
Laatste  Meting  2018-03-29 12:56:58+02:00

Hans-Nouwens, 2018-04-01:
Just er is iets niet juist in de tijdsaanduiding:
Bij een actuele tijd van 17:03 geven de units 17:03 +2 uur aan

justb4, 2018-04-01:
Tijdaanduiding is juist: 17:03 +2 uur betekent: 17:03 met een offset +2 van UTC/GMT. Dus UTC is dan 15:03. 17:03 is "onze" NL tijd. Die +2 is niets meer dan een aanduiding, net als CET etc, alleen exacter omdat o.a Daylight Savings wordt meegenomen.

Hans-Nouwens, 2018-04-01:
Hoewel volgens ISO een juiste beschrijving is, is het voor niet ingewijden onlogisch gezien de reacties hierop.
Dat het UTC +2 is, is onduidelijk omdat UTC nergens staat.

Ik vraag me af: Wat is het belangrijkste: volgens de regels juist of begrijpelijk voor de gebruiker?

Mij maakt het verder niet uit. Het is alleen verwarrend. Mocht je het niet aan kunnen/willen passen, dan laten we het zo.

justb4, 2018-04-03:
Laat maar even open, op SCLL heb ik geen budget momenteel, als ze er in ["SE Nijmegen Project" ](https://github.com/Geonovum/smartemission/projects/1) last van hebben kan het mogelijk via dat project.

justb4, 2018-04-10:
Bleek ook lastig bij PDOK migratie, dus aangepast. SmartApp herladen. Tijd moet iets geven als: 
`10-4-2018 om 15:06:24 NL`.

closed 2018-04-10 by justb4

Hans-Nouwens, 2018-04-10:
Hei Just,

 

Laatste meting is nu leeg

Hans-Nouwens, 2018-04-10:
> hmm, lijkt iets browser-specifieks te zijn. Chrome is ok. Safari niet. Welke browser/versie gebruik je?

Safari

Met groet, Hans Nouwens

justb4, 2018-04-10:
Zou nu goed moeten zijn, althans in Chrome+Safari+Firefox ok, IE heb ik niet.

Hans-Nouwens, 2018-04-10:
Chrome en Safari geven bij mij nog geen waarde bij laatste meting

justb4, 2018-04-10:
vreemd, browser caching? shift reload.

Hans-Nouwens, 2018-04-10:
Werkt!