smartemission/smartemission #111: upgrade GOST database

justb4, 2018-04-13:
the current master version of the GOST server contains performance enhancements. This includes modifications to the DB schema. See also https://github.com/gost/gost-db/issues/5

At least the following changes were detected:

```
ALTER TABLE v1.featureofinterest ADD COLUMN geojson jsonb;
ALTER TABLE v1.location ADD COLUMN geojson jsonb;

ALTER TABLE v1.observation DROP CONSTRAINT observation_pkey;
CREATE INDEX i_dsid_id
  ON v1.observation
  USING btree
  (stream_id, id);
```

But possibly we need to fill the added `geojson` fields from the `feature` `geometry` columns.

justb4, 2018-04-13:
Ok, appears that only new records have the `geojson` field populated:

[image]

dschowta, 2018-04-16:
Hi,

This might be off the topic, but I am one step behind this issue. 
I am getting following error when I upgrade GOST DB from 5 to 6.

> gost-db_1    | 2018-04-16 12:12:12.281 UTC [1] FATAL:  database files are incompatible with server
gost-db_1    | 2018-04-16 12:12:12.281 UTC [1] DETAIL:  The data directory was initialized by PostgreSQL version 9.6, which is not compatible with this version 10.3.
gost_gost-db_1 exited with code 1

Do you know easier way to upgrade my data? 
[I came across the tool pg_upgrade. But that is not an easy way for docker based  GOST DBs ]
Thanks in advance

justb4, 2018-05-29:
Done for all deploys (test, prod, pdok), see upgrade script: 
https://github.com/smartemission/smartemission/blob/master/database/upgrades/upgrade3.sql

or directly:

```
--
-- Upgrade GOST DB 0.5 --> 0.6
-- Just van den Broecke - april 17, 2018
--

-- NEW geojson shadow column for feature column
-- geojson column must be like {"type": "Point", "coordinates": [5.671321, 51.472583]}
ALTER TABLE v1.featureofinterest ADD COLUMN geojson jsonb;
UPDATE v1.featureofinterest SET geojson = concat('{"type": "Point", "coordinates": [' || ST_X(feature) || ', ' ||  ST_Y(feature) || ']}')::jsonb;

-- NEW geojson shadow column for feature column
ALTER TABLE v1.location ADD COLUMN geojson jsonb;
UPDATE v1.location SET geojson = concat('{"type": "Point", "coordinates": [' || ST_X(location) || ', ' ||  ST_Y(location) || ']}')::jsonb;

-- Optimizations for observation table
ALTER TABLE v1.observation DROP CONSTRAINT observation_pkey;
CREATE INDEX i_dsid_id
  ON v1.observation
  USING btree
  (stream_id, id);
```

May be of use for gost/gost-db#5

closed 2018-05-29 by justb4