smartemission/smartemission #112: Sensor not showing data on the SmartApp

Nijmegen-Geluid, 2018-04-18:
The sensor lights orange and flashes every 11 seconds white.
But no data is seen on the SmartApp.
Is it possible to send an mail or SMS to the citizen? So that the citizen can reset the sensor.

justb4, 2018-04-18:
First: which sensor number?

We currently have no registration of emails coupled to sensor numbers. At least the platform does not have this information. Once we have this info we can setup some kind of "uptime" checker.

Nijmegen-Geluid, 2018-04-20:
Okay. Very good. We make a list.
Are there any costs?

justb4, 2018-08-03:
Yes offcourse there are costs. Depends a bit on the solution:

- simple: have an external uptimechecker per sensor like GeoHealthCheck or UptimeRobot with email-address from user
- full: maintain users and sensors in platform with notifications

Please open separate issue with title like "Notify sensor-user when sensor data is not updated"

closed 2018-08-03 by justb4