smartemission/smartemission #121: Downloaden van data via site (heron) faalt, Alleen toegang tot metadata via API

dfnk, 2018-06-05:
Ik probeer de historische data van het smartemissions project te downloaden.

Via de Heron viewer krijg ik de foutmelding 'Not Found
The requested URL was not found on the server. If you entered the URL manually please check your spelling and try again.' nadat ik op downloaden druk.

Via de API krijg ik alleen metadata binnen en geen meetgegevens.

justb4, 2018-06-05:
Er waren idd een paar problemen, had o.a. te maken met migratie PDOK, waardoor URLs veranderd waren naar https, alleen dat moest niet voor de `WFS NameSpaces` gelden. 
Ook sommige proxy/static paden werkten niet (`components.json`).

Heb de versie op
http://data.smartemission.nl/heron gewijzigd, de download via verrekijker werkte daarna weer voor mij (wel Heron viewer herladen ivm browser caching). 

Definitieve fix nog te maken dus laat deze issue maar open ook al werkt het weer voor jou.

justb4, 2018-06-07:
@Sjoerd39 nieuwe versie Heron viewer draait nu. kun je checken en issue sluiten als 't werkt?

closed 2018-06-07 by Sjoerd39