smartemission/smartemission #129: Matching van meetwaarden en meta data via device id's

Sjoerd39, 2018-06-19:
Zijn de meetwaarden in de database aan de juiste devices gekoppeld? Het lijkt erop dat er een aantal fouten in de data zit wat betreft het juist toewijzen van een device id aan meetwaarden of andersom. 

Wanneer ik de data voor de sensoren in Nijmegen ophaal en wil koppelen aan metadata (long, lat etc. via ) zijn er een aantal meetstations met meetwaarden van een verkeerd device. Daarbij zijn er device id's met data die geen metadata hebben en vice versa. 

Voorbeeld: voor device id 16 heb ik wel metadata, maar geen meetwaarde. Als ik in de heron viewer kijk naar device 16 (groene driehoek 16) zie ik in de meetwaarden vooral device id  58, maar ook 70,100 en 103 terugkomen. maar bij de metadata staat wel device id nr 16.

Ander voorbeeld: er zijn 9936 meetwaarde toegekend aan device id 15, maar heb ik geen meta data voor device id 15 en zie 'm niet in de kaart staan.

justb4, 2018-06-19:
Kun je specifieker zijn: als je een issue opent, beschrijf dan de specifieke stappen, bijv protocol requests of Heron akties die je uitvoert, het resultaat dat je verwacht en het resultaat dat je ziet. Dan  kan degene die issue oplost deze nalopen. Zoals hierboven beschreven is dit te vaag.

Bijv

* *"Wanneer ik de data voor de sensoren in Nijmegen ophaal"*: hoe? via SOS, WFS, Heron, etc?
* *"wil koppelen aan metadata"*": welke metadata, hoe verkregen?
* "*zie ik in de meetwaarden vooral device id 58*": welke lagen/meetwaarden in Heron zijn aangezet, welke akties tot dit resultaat?

etc

Wat er wel aan de hand is:

* vorige week is de server gecrashed en alle data verdwenen, deze (de historie) wordt nu opnieuw opgebouwd vanuit de data-collectors van Intemo en CityGIS , dat zal nog zeker 2 weken duren. 
* de groene driehoeken in Heron laten de huidige station-locaties zien
* sensoren zijn in de historie verplaatst en hernummerd
* bij historische lagen zie je de waarden (cirkels) van de waarde van station op die locatie op dat tijdstip. Dat kan anders zijn dan huidige stations-locatie (bijv waar nu 16 is, was blijkbaar eerder 58)

Uit nieuwsgierigheid, kan per email naar mij: voor wie/welk project werk je?

justb4, 2018-06-22:
Het lijkt erop dat niet alle data direct terug te halen is: een serie stations is verhuisd van de CityGIS naar de Intemo Data Collector in feb/mrt 2018. Het gaat om de sensoren: `22,29,36,37,49,53,67,82,83,85,89,92,102,126,133`. De data van voor die tijd is momenteel niet aanwezig via de APIs. Dit zal extra effort kosten om terug te halen, ik zal dit met project opnemen.

Sjoerd39, 2018-06-26:
Is er een overzicht van de verplaatsingen en hernummering van devices beschikbaar?

Sjoerd39, 2018-06-26:
De vraag is dus: Hoe krijg ik die totale tabel, met alle meetwaarden en alle metagegevens boven tafel? 

Om het even samen te vatten, mijn doel is een compleet overzicht van alle meetwaarden van de (eerdere) meetstations in Nijmegen te krijgen met de metadata als coördinaten waar de meetstations hangen (of gehangen hebben). Maar ik zie nog geen mogelijkheid dit via de 'Heron manier' uit de database te halen.

Voorbeeld: Als ik via de Heron Viewer de historische data van meetstation 16 in Nijmegen opzoek via: 'verrekijker --> zoeken door tekenen van geometrie-->  selectielijst historische meetwaarden: temperatuur --> en ik teken een cirkel rondom station 16' dan is het resultaat een tabel met meetwaarden van (eerdere) meetstations met andere device_id's. Hierbij zitten geen meetwaarde van station 16 (device_id 16). Ik krijg via Heron de metadata van station 16 wel boven tafel, maar geen metadata van de andere stations in de tabel.

justb4, 2018-06-26:
@Sjoerd39 Ok, dus "metadata=coordinaten"? Nogmaals: er zijn een aantal dingen aan de hand:

* de data wordt momenteel opnieuw opgebouwd, veel (historische) data ontbreekt nog, werken we aan
* "zoeken door tekenen" is niet geheel stabiel, beter is "Download Historie/Tijdreeksen"
* als je een lijst met sensor-nummers hebt kun je per sensor-nr de historie downloaden
* download als CSV in Heron geeft geen coordinaten, wel bijv GeoJSON of GML of direct via WFS (zie onder)

Punt is: er zitten straks weer plm 8 miljoen uurwaarden in de database. Elk wil daaruit een eigen selectie of manier van selecteren. Via WFS is in feite alles mogelijk (dat is wat Heron onder water gebruikt): op sensor nummer, coordinaten, periode, of gecombineerd. Ik denk dat in jouw geval een aantal specifieke WFS queries het beste zou werken. Bijv vanaf commandline of in browser alle NO2 van station 16:

http://data.smartemission.nl/geoserver/wfs?request=GetFeature&typename=timeseries_no2&cql_filter=device_id=16&outputformat=csv

Hier zitten ook coordinaten in. Zie ook [de documentatie, WFS onderaan](http://data.smartemission.nl/data):

  "*For raw/direct download via your browser or a download tool like curl, a quick WFS download be done this base URL:*

  http://data.smartemission.nl/geoserver/wfs?request=GetFeature&typename=timeseries_component&outputformat=csv&CQL_FILTER=device_id=station id

  *where component is one of 'co2', 'co', 'coraw', 'no2', 'no2raw', 'o3', 'o3raw', 'temperature',*
  *'pressure', 'humidity', 'noise_level_avg' and station id a station number like 12."*

Wel even geduld...zijn flinke hoeveelheden data.

justb4, 2018-06-26:
@Sjoerd39 @avdcruyssen Die "geofence" is juist wat WFS mogelijk maakt, een zgn "spatial query" (ruimtelijk zoeken), bijv alle metingen in een bounding box of polygoon. Er zit vaak wel een limiet (aantal records) wat de WFS teruggeeft (nu 1000000).  Voorbeeld alle PM10 metingen in rechthoek bij Waalkade:

http://data.smartemission.nl/geoserver/wfs?request=GetFeature&typenames=timeseries_pm10&outputformat=csv&BBOX=5.858,51.8,5.874,51.9&version=1.0.0&srsName=EPSG:4326

justb4, 2018-07-04:
Let wel: de data wordt momenteel op andere server opnieuw opgebouwd en later naar de productie-server overgezet. Dit omdat met name historie plm van voor 2018 voor sensoren: `22,29,36,37,49,53,67,82,83,85,89,92,102,126,133` (deels) ontbreekt. Ook is m.n. de geluids berekening medio 2017 verbeterd. Als het alleen om waarden na plm maart 2018 gaat en alle "niet-Nijmegen-sensoren" (Smart City Living Lab, i.e. 2000-range), voldoet huidige dataset/historie geheel.