smartemission/smartemission #153: Inserting new sensor of other brand

Nijmegen-Geluid, 2018-12-18:
Een fijnstofmeter kit van een burger heeft de mogelijkheid om de data naar een eigen API te versturen. Is gebaseerd op Luftdaten!
Kan dit binnen het SE projekt?

justb4, 2018-12-20:
Alles is mogelijk, alleen niet out-of-the box. Er is een klein stukje ontwikkeling nodig, m.n. voor de mapping van data formaten. Als deze naar InfluxDB kan sturen zijn we een eind op weg.

Nijmegen-Geluid, 2018-12-21:
Is mogelijk.
Via een output in de software.
Hoe krijgen we dat bij jouw?
Als extern bestandje via de mail?

Nijmegen-Geluid, 2018-12-23:
Je kunt op https://github.com/opendata-stuttgart/sensors-software alle info vinden. Als je het account geeft waar het bericht heen moet kan ik dit bij een van de sensors activeren.
Is dat een goed idee?

justb4, 2019-01-24:
Nog niet vergeten! Ga dit oppakken, ga zo naar dit event: https://www.meetup.com/Code-For-NL/events/258154926 om meer over Luftdaten (sensors) te weten te komen.

Nijmegen-Geluid, 2019-01-24:
Perfect.

Ik ben bezig met een goedkope geluidsensor.
En ik heb iemand die al een goed werkende fijnstofmeter heeft.
Wil die graag in het SE net hangen.

Veel plezier bij de meeting. Ik ben volgende week donderdag weer terug uit de Dolomieten.

Gr. Peter

Met vriendelijke groet,

Peter van der Voorn
Specialist Geluid
Bureau Leefomgevingskwaliteit PK40
Team Geluid & Lucht
Postbus 9105, 6500 HG, Nijmegen<x-apple-data-detectors://6/1>
Korte Nieuwstraat 6, 6511 PP, Nijmegen<x-apple-data-detectors://7/0>
M: 06 51 33 07 03<tel:06%2051%2033%2007%2003>
T: 024 329 96 92<tel:024%20329%2096%2092>
E: p.van.der.voorn@nijmegen.nl<mailto:p.van.der.voorn@nijmegen.nl>
W: www.nijmegen.nl<http://www.nijmegen.nl/>
[logo nijmegen]

Nijmegen-Geluid, 2019-01-28:
Hallo Just,

Degene die zelf fijnstof sensoren maakt in Nijmegen (en die ik in SE zou willen hangen) werkt 4 dagen per week bij het Kadaster in Apeldoorn bij IT (locatie de Brug).

Kun je rechtstreeks met hem contact opnemen?

Zijn naam is Roderick Peters. Zijn mobiele nummer is +31 6 10 89 75 45
Mail roderick.peters@telfort.nl<mailto:roderick.peters@telfort.nl>

Gr. Peter

Met vriendelijke groet,

Peter van der Voorn
Specialist Geluid
Bureau Leefomgevingskwaliteit PK40
Team Geluid & Lucht
Postbus 9105, 6500 HG, Nijmegen<x-apple-data-detectors://6/1>
Korte Nieuwstraat 6, 6511 PP, Nijmegen<x-apple-data-detectors://7/0>
M: 06 51 33 07 03<tel:06%2051%2033%2007%2003>
T: 024 329 96 92<tel:024%20329%2096%2092>
E: p.van.der.voorn@nijmegen.nl<mailto:p.van.der.voorn@nijmegen.nl>
W: www.nijmegen.nl<http://www.nijmegen.nl/>
[logo nijmegen]

justb4, 2019-01-29:
Ik stel voor dat Roderick hier reageert, dan blijft de informatie uitwisseling transparant, ipv ergens 1-1 in emails. Ik kan hem wel vragen.

Ik ben redelijk, ook technisch, op de hoogte van LuftDaten. Het beste, ook voor toekomst is als het SE platform de data van LuftDaten zelf binnenhaalt via onze "Harvesters". Dit is niet anders dan we met Intemo en AirSensEUR doen: data wordt op remote server, een Data Collector, verzamelt en SE haalt deze voortdurend op. Dan kunnen we ook snel uitbreiden met meerdere LuftDaten sensors en is het gemakkelijk voor de LuftDaten sensors eigenaren: hoeven maar naar 1 plek te sturen. Deze Harvester moet nog wel ontwikkeld worden, maar er hoeft verder geen calibratie gedaan neem ik aan, hoewel die SDS011 wel erg gevoelig is voor RH, maar goed er worden meerdere sensoren in LuftDaten gebruikt.

Nijmegen-Geluid, 2019-01-29:
Oké. Bij deze.
Transparant wil ik het graag zien.
Roderick: pak jij dit verder op met Just.
En houden jullie mij op de hoogte via deze mailinglist.

Graag de Harvester voor de data ontwikkelen.
Volgens mij heeft Roderick alles pasklaar staan!?

Dank jullie wel vast!

Gr. Peter

Met vriendelijke groet,

Peter van der Voorn
Specialist Geluid
Bureau Leefomgevingskwaliteit PK40
Team Geluid & Lucht
Postbus 9105, 6500 HG, Nijmegen<x-apple-data-detectors://6/1>
Korte Nieuwstraat 6, 6511 PP, Nijmegen<x-apple-data-detectors://7/0>
M: 06 51 33 07 03<tel:06%2051%2033%2007%2003>
T: 024 329 96 92<tel:024%20329%2096%2092>
E: p.van.der.voorn@nijmegen.nl<mailto:p.van.der.voorn@nijmegen.nl>
W: www.nijmegen.nl<http://www.nijmegen.nl/>
[logo nijmegen]

ropeters68, 2019-01-30:
Hallo Just,

Ik zie 2 manieren om de luftdaten sensor data beschikbaar te maken voor het Smart Emission platform: 

- data ophalen luftdaten http://api.luftdaten.info/v1/sensor/{sensorid} (voorbeelden: sensor Extrmadura straat Lent 16772 (SDS011), 16773 (DHT), 16774 (BME280)) 
Randvoorwaarde: de sensors moeten bij lufdaten aangemeld zijn.
Hierbij moet er rekening gehouden worden dat luftdaten per sensor in de kit een ID gebruikt, onze kits hebben een SDS011 voor fijnstof, een DHT22 bij de luchtafvoer (voor controle op de temperatuur/ luchtvochtigheid van de gemeten lucht), BME280 (meet de temperatuur/luchtvochtigheid en luchtdruk van de buitenlucht). Onze sensors wijken iets af van de basis kits zoals op de Luftdaten site beschreven. Wij gebruiken een verwarmde luchtinlaat (en SDS011) behuizing) waardoor de te meten lucht ca.  10 graden wordt verwarmd waardoor de relatieve luchtvochtigheid altijd lager zou moeten liggen dan 70%. De bedoeling is dat daarmee de meetfout door hygroscopische groei van de fijnstofdeeltjes wordt ondervangen. 
De kits versturen iedere 150 seconden een bericht naar luftdaten (op het configuratiescherm kan een andere frequentie ingesteld worden maar daar doen we tot nu toe nog niets mee).
- Data via een eigen API versturen
Verder heb ik een gratis accountje voor het opvragen van weersgegevens:
http://weerlive.nl/api/json-data-10min.php?key=xxxxxxxx&locatie=Nijmegen
Kan interessant zijn voor ivm verklaren van veranderingen agv wijziging in windrichting/snelheid.
De eerste optie lijkt me het eenvoudigst, is dan een kwestie van de json-file mappen naar de database.

ropeters68, 2019-01-30:
Kleine toevoeging, naar verwachting worden er in februari in workshop in Nijmegen/Arnhem ca. 40 sensors gebouwd, dit zijn wel de basis versies (SDS011 en DHT22).

justb4, 2019-02-04:
@ropeters68 ja via Harvesting, op selectie van Luftdata sensoren, evt geografisch regio Nijmegen, indien mogelijk, lijkt mij beste optie. Zie idd 2 metingen uit laatste 5 min: bijv http://api.luftdaten.info/v1/sensor/16772/  (slash '/' verplicht aan eind URL). DIe workshop lijkt mij interessant. Worden die sensoren daar ook meteen aangesloten? Luftdaten sensor id kan als metadata in SE.

Ja RH is m.i. groot probleem en DHT22 blijft snel op 100% hangen.  

Ik overleg nog met Paul Geurts, maar lijkt mij verrijking voor SE platform.

justb4, 2019-02-05:
Goed/prio met Paul.
Ja, Lukas is druk, heb hem uitgebreid gesproken vorige week vr op TTN conf. Mogelijk iets met SE platform doen tbv Luftdaten, maar stap voor stap. 

WIFI en Stroom zijn ook in mijn AQ-projecten (heb i.s.m. RIVM 5 EU JRC AirSensEURs aangesloten op SE-platform, o.a. Nijmegen-Ruyterstr, steeds grootste struikelblok. Daarom ben ik mij aan verdiepen in LoRa/TTN evt. later met solar-current. Maar die AQ sensoren hebben wel vaak een actieve ventilator die veel stroom vreet.

De BME280 hoor ik overal goede verhalen over. 

Terug naar issue hier: Ik ga kijken hoe op slimste manier LuftDaten (LD) sensoren kunnen worden geharvest, m.n. zonder handmatige akties. Bijv alles in bbox "Groot-Nijmegen". LD API heeft idd spatial queries, bijv bbox of lon/lat met straal, daar kom ik wel uit.  Ik wil wel de LD sensor-id's behouden, anders dan bijv in https://samenmeten.rivm.nl/dataportaal/. Gelukkig heeft SE een generieke "metadata" mogelijkheid. Voornaamste probleem is een automatische "id-mapping":  binnen SE zijn id's iets anders opgebouwd, een id is een fysiek Station ("Thing" in SensorThings API) : 8-cijferig, is historisch ontstaan i.s.m. Intemo: eerste 4 cijfers project id/vaak zelfs postcode, tweede 4 station-id. Binnen "Station" kunnen weer meerdere sensors/indicatoren aktief zijn tot wel 60 voor Josene (denk aan geluid-indicatoren in freq banden, sommige "sensors" zijn virtueel, afgeleid). Goed, er is altijd uit te komen. Ik laat weten. Ik begin met laatste waarden, is eenvoud, die zijn altijd in [SmartApp](https://data.smartemission.nl/smartapp/) en [SOSEmu API](https://data.smartemission.nl/sosemu/) beschikbaar.

justb4, 2019-02-06:
Ok, klopte niet wat ik over LD en id's en SamenMeten- hierboven stelde: er is toch een concept van een "station" in LD, dat is een "kit" bijv met `id=LTD_8930` (Lent) zie ik terug in de LD API als een `location.id` waarin hier 3 sensoren (SDS011, SHT22, BME280), dus totaal komt ie 6x terug, hetgeen goed klopt met dat ieder station iedere 150 sec stuurt (dus 2x per 5 min).

Met deze query krijg ik alle laatste 5 min metingen in "Groter Nijmegen" gebied:
http://api.luftdaten.info/v1/filter/box=51.7,5.6,51.9,6.0

```
.
.
  {
    "sensor": {
      "pin": "1",
      "id": 17622,
      "sensor_type": {
        "name": "SDS011",
        "id": 14,
        "manufacturer": "Nova Fitness"
      }
    },
    "timestamp": "2019-02-05 15:42:03",
    "location": {
      "latitude": "51.8600",
      "country": "NL",
      "id": 8930,
      "longitude": "5.8680",
      "altitude": "12.7"
    },
    "id": 2826167868,
    "sampling_rate": null,
    "sensordatavalues": [
      {
        "value_type": "P1",
        "id": 6003944572,
        "value": "30.00"
      },
      {
        "value_type": "P2",
        "id": 6003944573,
        "value": "23.50"
      }
    ]
  },
  {
    "sensor": {
      "pin": "7",
      "id": 17623,
      "sensor_type": {
        "name": "DHT22",
        "id": 9,
        "manufacturer": "various"
      }
    },
    "timestamp": "2019-02-05 15:42:04",
    "location": {
      "latitude": "51.8600",
      "country": "NL",
      "id": 8930,
      "longitude": "5.8680",
      "altitude": "12.7"
    },
    "id": 2826167908,
    "sampling_rate": null,
    "sensordatavalues": [
      {
        "value_type": "humidity",
        "id": 6003944658,
        "value": "47.80"
      },
      {
        "value_type": "temperature",
        "id": 6003944657,
        "value": "16.00"
      }
    ]
  },
```

justb4, 2019-02-06:
Eerste resultaten laatste waarden Luftdaten in Lent bemoedigend (test server, nog niet live), nu formaliseren en historische waarden bewaren.
 
[image]

justb4, 2019-02-06:
Eerste ruwe versie op productie:

* harvesting configurabel, kan per bbox, [nu allen Groter Nijmegen](https://github.com/smartemission/docker-se-stetl/blob/master/config/last.cfg#L54)
* alleen laatste waarden
* SE project id is `4931` (49 voor Luftdaten algemeen, 31 NL)
* eerste 4 cijfers station id is project id, rest Luftdaten kit id (location id): e.g. `49318526`
* te zien in SmartApp https://data.smartemission.nl/smartapp/ (ook lagen aan/uit te zetten)
* ook in Heron Viewer: (alleen "current" lagen):  https://data.smartemission.nl/heron/ 
* data via Harvester in SE DB
* nog dubbele waarden meteo (SHT22 en BME260)
* eerst dit testen dan verder met historie harvesting

[image]

ropeters68, 2019-02-07:
Top! Ziet er goed uit.

justb4, 2019-02-12:
Also integrated a [Luftdaten Harvester](https://github.com/smartemission/docker-se-stetl/blob/master/config/harvester_luftdaten.cfg). Every hour the average hour-values for Luftdaten stations within specified bboxes are read from the[ Luftdaten API.](https://github.com/opendata-stuttgart/meta/wiki/APIs)

Now historic time-series are also built up and made available via APIs (WMS, WFS, SOS, STA) and via the Heron Viewer (below). Will take some time for history to build up. Also some changes to metadata such that it is easier to track back a measurement to an LTD station and sensor (id) within. 

Made it easier to integrate additional sensor-types that send calibrated data via the introduction of the Vanilla Device. See also updated documentation:
https://smartplatform.readthedocs.io/en/latest/sensors.html

[image]

For now this implementation is finished and goes into maintenance mode so closing this issue. Open separate issues for any problems or extensions.

closed 2019-02-12 by justb4