smartemission/smartemission #155: Data Whaleserver

Nijmegen-Geluid, 2019-02-21:
The data on the Whale viewer is saved.
Can the data be used in the future?

justb4, 2019-02-23:
I don't fully understand the question. The situation is as follows:

* harvesting from the CityGIS Whale server is phased out
* all raw data (since about 2015) has been saved within the Smart Emission server database running now at PDOK/Kadaster
* raw values for all indicators (sensors) are stored as JSON "blobs" per station/per hour
* there is currently not an easy way to download this data, e.g. from within a browser

Attached is an example how these JSON blobs look like. The Josene manual is required to decode these values or the SE code can be used, like [here for audio indicators](https://github.com/smartemission/docker-se-stetl/blob/master/smartem/devices/josenefuncs.py#L177), basically 3 bands are encoded in each integer value:

```
# decode dB(A) values into 3 bands (0,1,2) for this octave
bands = [float(input_value & 255), float((input_value >> 8) & 255), float((input_value >> 16) & 255)]
```

[jose1.json.txt](https://github.com/smartemission/smartemission/files/2896758/jose1.json.txt)

Nijmegen-Geluid, 2019-02-23:
The data is still there.
Good to know.

When I want to see the noise data over the past three years I can see the data in a viewer?

Met vriendelijke groet,

Peter van der Voorn
Specialist Geluid
Bureau Leefomgevingskwaliteit PK40
Team Geluid & Lucht
Postbus 9105, 6500 HG, Nijmegen<x-apple-data-detectors://6/1>
Korte Nieuwstraat 6, 6511 PP, Nijmegen<x-apple-data-detectors://7/0>
M: 06 51 33 07 03<tel:06%2051%2033%2007%2003>
T: 024 329 96 92<tel:024%20329%2096%2092>
E: p.van.der.voorn@nijmegen.nl<mailto:p.van.der.voorn@nijmegen.nl>
W: www.nijmegen.nl<http://www.nijmegen.nl/>
[logo nijmegen]

justb4, 2019-02-24:
No, that has to be developed: both a download service and a viewer for raw data.

Nijmegen-Geluid, 2019-02-25:
Okay.
I ask Paul Geurts to get an order for you.
What are the costs you think?

Gr. Peter

justb4, 2019-02-25:
Best we discuss costs/estimates offline. 

@geurp0 @Nijmegen-Geluid I think that we should aim for a "Smart Emission Data Scientist's App" which allows specialized users like citizen scientists to browse, select, download and view any raw and refined (calibrated) data from the SE platform. We can build this step by step. I think this is a better approach than developing ad-hoc viewers. We need these as well, but we have APIs for that on which they can be build.